PRBonn/kiss-icp #3: Output result format

rlabs-oss, 2022-10-04:
Please give a hint as to the fields in the ..txt output file contained in the results folder when processing .xyz dataset.
Thank you.

nachovizzo, 2022-10-04:
Hello @rlabs-oss sorry for not printing a more detailed output on the CLI interface. We use the kitti-format, since we think it is quite standard. If you want to visualize the trajectory, you could use the evo tools for example(`pip install evo) and run the following command to see the output:

```sh
evo_traj kitti results/latest/*.txt -p
```

As an example, after running sequence `07` of the KITTI dataset and plotting the results I obtained this:
[image]

rlabs-oss, 2022-10-04:
Thank you - much appreciated.

closed 2022-10-04 by rlabs-oss