PRBonn/kiss-icp #6: Ros2 bag support ?

doisyg, 2022-10-06:
Hello,
Paper looks great! Is there a way to test this with ros2 bags ?

nachovizzo, 2022-10-06:
Thanks!

You can test it with ROS1 bagfiles, I haven't treid with ROS2 so far, but it should somehow work. in any case, the rosbag support is experimental, and it's also super slow, reading from the bagfile takes more time than running the pipeline.

In any case, if you just want to inspect the results, you can try to extract the pointclouds from the rosbag and then proces the folder. In ROS1 this can be easily done with `rosrun pcl_ros bag_to_pcd <bag.bag> <topic_name> <folder_output>`

We also plan to release a ROS1 and ROS2 packages soon-ish.

I hope this helps!

PS: Do you have access to a ROS2 bagfile you can upload or send a Link? I can try to see if there is an easy way to process this.

doisyg, 2022-10-06:
That would be amazing. 
Just sent you an wetransfer link on your email

nachovizzo, 2022-10-14:
@doisyg Sorry for the delay, I made a “preliminary” in a rush dataloader for rosbag2 files. It is working at least with the rosbag file that you sent me, though the results are not great. I just pushed a quick fix only for linux, if you are using other platform please let me know. Please `pip install -U kiss-icp` to upgrade the kiss-icp version

I ran this command to process your bagfile:
```sh
kiss_icp_pipeline --dataloader rosbag2 $DATASETS/github/rosbag2_horizontal --topic /rslidar_points --visualize
```
Additionally, I change the default max range of 100m to 10m since this looks like an indoor since (though can't really tell by just looking the data). You could extend the dataloader to apply any transformation that put the point cloud upside down (or whatever pre-processing you need)

```yaml
data:
  max_range: 10.0
  min_range: 0.0
```

[image]

Please let me know if it worked on your end!

doisyg, 2022-10-16:
Amazing, thanks, it works!
I had to install `pip install rosbags` in addition de open3d.

nachovizzo, 2022-10-16:
Great! We will update the requirements once the final release is out there

closed 2022-10-16 by nachovizzo

nachovizzo, 2023-01-12:
@jeychandar if you successfully installed the package `pip install kiss-icp`, then you can process a ROS2 bagile by running:

```sh
$ kiss_icp_pipeline --dataloader rosbag2 <dataset_path> --topic <pc2 topic> --visualize
```

jeychandar, 2023-01-12:
Did the following command but shows error. 
[image]

nachovizzo, 2023-01-12:
Can you share the cmmand? You are invoking the ROS1 dataloader there, not the ROS2

jeychandar, 2023-01-12:
kiss_icp_pipeline --dataloader rosbag2 /home/praveen/Downloads/kitti.bag --topic imu_raw --visualize

nachovizzo, 2023-01-12:
But `Kitti.bag` then is a ROS1 bag, try with the `rosbag` dataloader instead

jeychandar, 2023-01-12:
Tried with rosbag also still getting the same error. I am trying play and see how the kitti bag works.I am able to play in rviz for some kind of sensor. velodyne, ouster,livox I am able to visualize and can store to pcd and convert to las. But your paper says you have used velocity model and downsampling voxel by find the ICP and fast nearest neighbour. I was impressed and wanna try these recorded bag with yours and see how the results are.

nachovizzo, 2023-01-12:
@jeychandar so, this is NOT a  ROS1 bagfile, I just downloaded and run:
```sh
$ kiss_icp_pipeline 2011_09_30_drive_0028.bag --topic /points_raw --visualize
``` 

It's working:
[image]

But I'm not sure why you want to use that particular bagfile, there is no `timestamp` information, which means that you won't be able to "deskew" the pointclouds. I recommend using the KITTI dataloader instead.

nachovizzo, 2023-01-12:
These are the poses obtained by KISS-ICP, **without** deskewing:

[image]
[2011_09_30_drive_0028_poses_kitti.txt](https://github.com/PRBonn/kiss-icp/files/10400773/2011_09_30_drive_0028_poses_kitti.txt)

**At this point, PLEASE open a new issue since this is not at all related with ROS2 support**

jeychandar, 2023-01-12:
> @jeychandar so, this is NOT a ROS1 bagfile, I just downloaded and run:
> 
> ```shell
> $ kiss_icp_pipeline 2011_09_30_drive_0028.bag --topic /points_raw --visualize
> ```
> 
> It's working: [image]
> 
> But I'm not sure why you want to use that particular bagfile, there is no `timestamp` information, which means that you won't be able to "deskew" the pointclouds. I recommend using the KITTI dataloader instead.

I am happy that it is working and thanks for sharing the output.The problem that I am facing is I am not able to visualize.I didn't noticed that this particular bag doesn't have timestamp sorry for that.I used kitti dataloder also but still showing the same error sir.