PRBonn/kiss-icp #8: Is it possible to use this library in a runtime algorithm?

sisaha9, 2022-10-08:
Basically use this to run in real time, taking pointclouds one by one and updating position (like through a ROS/ROS2 system)

nachovizzo, 2022-10-10:
Hello @sisaha9 thanks for the question. I guess that what you mean is that if it is possible to use the library to run at sensor frame-rate, also **wrongly** named "realtime".

The answer is YES, but it depends on the resources available of your system. On my old laptop it runs up to `70 Hz` for the KITTI dataset, but if you are deploying this on a mobile robot with resource constrains the runtime of KISS-ICP will be not as fast as on a laptop.

closed 2022-10-10 by nachovizzo

sisaha9, 2022-10-10:
@nachovizzo Is there documentation on using this at sensor frame rate? That is passing in pointclouds as they come in and using that to update the car's position?

nachovizzo, 2022-10-12:
I'm sorry, but I can't understand the question. There is no such thing as "using this at sensor frame rate". If you are using a robot, you just need to use the KISS-ICP library to process the pointclouds

lhy0807, 2023-01-14:
> I'm sorry, but I can't understand the question. There is no such thing as "using this at sensor frame rate". If you are using a robot, you just need to use the KISS-ICP library to process the pointclouds

Thanks for the awesome work! I also have a similar question: Is it possible to deploy KISS-ICP in a real-time ROS system? From the README, it seems that it currently could only process datasets.

nachovizzo, 2023-01-15:
We are working on a ROS wrapper, stayed tuned ;)