PRBonn/kiss-icp #9: Error when using Kitti data_odometry_gray dataset

espackman, 2022-10-09:
Hi!

I am attempting to run KISS ICP on the Kitti gray odometry dataset.

I am using this argument 
`kiss_icp_pipeline --dataloader kitti --sequence 07 --visualize "path\to\dataset"`
which results in this error:
`┌─────────────────────────────── Traceback (most recent call last) ────────────────────────────────┐
│ C:\ProgramData\Anaconda3\lib\site-packages\kiss_icp\tools\cmd.py:144 in kiss_icp_pipeline        │
│                                                                                                  │
│   141 │   from kiss_icp.datasets import dataset_factory                                          │
│   142 │   from kiss_icp.pipeline import OdometryPipeline                                         │
│   143 │                                                                                          │
│ > 144 │   OdometryPipeline(                                                                      │
│   145 │   │   dataset=dataset_factory(                                                           │
│   146 │   │   │   dataloader=dataloader,                                                         │
│   147 │   │   │   data_dir=data,                                                                 │
│                                                                                                  │
│ ┌─────────────────────────────────────────── locals ───────────────────────────────────────────┐ │
│ │           config = WindowsPath('C:/ProgramData/Anaconda3/lib/site-packages/kiss_icp/config/… │ │
│ │             data = WindowsPath('C:/Users/Everett/KITTI/dataset')                              │ │
│ │       dataloader = 'kitti'                                                                   │ │
│ │  dataset_factory = <function dataset_factory at 0x000001FC120119D0>                          │ │
│ │           deskew = False                                                                     │ │
│ │             jump = 0                                                                         │ │
│ │        max_range = None                                                                      │ │
│ │          n_scans = -1                                                                        │ │
│ │         no_cache = None                                                                      │ │
│ │ OdometryPipeline = <class 'kiss_icp.pipeline.OdometryPipeline'>                              │ │
│ │         sequence = 7                                                                         │ │
│ │            topic = None                                                                      │ │
│ │        visualize = True                                                                      │ │
│ └──────────────────────────────────────────────────────────────────────────────────────────────┘ │
│                                                                                                  │
│ C:\ProgramData\Anaconda3\lib\site-packages\kiss_icp\pipeline.py:50 in run                        │
│                                                                                                  │
│    47 │   # Public interface  ------                                                             │
│    48 │   def run(self):                                                                         │
│    49 │   │   self._run_pipeline()                                                               │
│ >  50 │   │   self._write_result_poses()                                                         │
│    51 │   │   self._write_cfg()                                                                  │
│    52 │   │   return self._run_evaluation()                                                      │
│    53                                                                                            │
│                                                                                                  │
│ ┌───────────────────────────────── locals ─────────────────────────────────┐                     │
│ │ self = <kiss_icp.pipeline.OdometryPipeline object at 0x000001FC13ECC1F0> │                     │
│ └──────────────────────────────────────────────────────────────────────────┘                     │
│                                                                                                  │
│ C:\ProgramData\Anaconda3\lib\site-packages\kiss_icp\pipeline.py:91 in _write_result_poses        │
│                                                                                                  │
│    88 │   │   return self._get_results_dir(self.config.out_dir)                                  │
│    89 │                                                                                          │
│    90 │   def _write_result_poses(self):                                                         │
│ >  91 │   │   self._write_poses(self.poses, f"{self.results_dir}/{self._dataset.sequence_id}.t   │
│    92 │                                                                                          │
│    93 │   def _get_fps(self):                                                                    │
│    94 │   │   total_time_ns = reduce(lambda a, b: a + b, self.times)                             │
│                                                                                                  │
│ ┌───────────────────────────────── locals ─────────────────────────────────┐                     │
│ │ self = <kiss_icp.pipeline.OdometryPipeline object at 0x000001FC13ECC1F0> │                     │
│ └──────────────────────────────────────────────────────────────────────────┘                     │
│                                                                                                  │
│ C:\ProgramData\Anaconda3\lib\site-packages\kiss_icp\pipeline.py:80 in _write_poses               │
│                                                                                                  │
│    77 │   │   np.savetxt(                                                                        │
│    78 │   │   │   fname=filename,                                                                │
│    79 │   │   │   X=_to_kitti_format(                                                            │
│ >  80 │   │   │   │   self._dataset.apply_calibration(poses)                                     │
│    81 │   │   │   │   if hasattr(self._dataset, "apply_calibration")                             │
│    82 │   │   │   │   else poses                                                                 │
[... 8 lines cut ...]
│ └──────────────────────────────────────────────────────────────────────────────────────────────┘ │
│                                                                                                  │
│ C:\ProgramData\Anaconda3\lib\site-packages\kiss_icp\datasets\kitti.py:60 in apply_calibration    │
│                                                                                                  │
│    57 │   │   """Converts from Velodyne to Camera Frame"""                                       │
│    58 │   │   Tr = np.eye(4, dtype=np.float64)                                                   │
│    59 │   │   Tr[:3, :4] = self.calibration["Tr"].reshape(3, 4)                                  │
│ >  60 │   │   return Tr @ poses @ np.linalg.inv(Tr)                                              │
│    61 │                                                                                          │
│    62 │   @memoize()                                                                             │
│    63 │   def read_point_cloud(self, scan_file: str, config: KISSConfig.data):                   │
│                                                                                                  │
│ ┌────────────────────────────────────── locals ───────────────────────────────────────┐          │
│ │ poses = []                                                                          │          │
│ │  self = <kiss_icp.datasets.kitti.KITTIOdometryDataset object at 0x000001FC127577F0> │          │
│ │    Tr = array([[-0.00185774, -0.99996597, -0.00803998, -0.00478403],                │          │
│ │         │      [-0.00648147,  0.00805186, -0.99994659, -0.07337429],                │          │
│ │         │      [ 0.99997729, -0.00180553, -0.0064962 , -0.3339968 ],                │          │
│ │         │      [ 0.        ,  0.        ,  0.        ,  1.        ]])               │          │
│ └─────────────────────────────────────────────────────────────────────────────────────┘          │
└──────────────────────────────────────────────────────────────────────────────────────────────────┘
ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature
(n?,k),(k,m?)->(n?,m?) (size 0 is different from 4)`

Is there any suggestions on how to resolve this issue?

nachovizzo, 2022-10-10:
Hello @espackman.

It is nearly impossible for me to read that log, but if I need to guess, you are missing the poses file `07.txt` for doing the evaluation. This is a bit tricky since everyone has it on different places.

The expected layout for us is
```sh
<data/kitti-odometry/dataset>
 ├── poses
 │   ├── 00.txt
 │   ├── 01.txt
 │   ├── 02.txt
 │   ├── 03.txt
 │   ├── 04.txt
 │   ├── 05.txt
 │   ├── 06.txt
 │   ├── 07.txt
 │   ├── 08.txt
 │   ├── 09.txt
 │   ├── 10.txt
 └── sequences
     ├── 00
     │   ├── calib.txt
     │   ├── poses.txt
     │   ├── times.txt
     │   └── velodyne
     ├── 01
     │   ├── ....
```

If this still does not work, are you able to run the pipeline until it finish? can you describe a bit more the issue? thanks!

espackman, 2022-10-11:
Hi @nachovizzo ,

Thanks for the advice! You are correct, I was missing the poses.txt from the sequences folder... I thought it would be enough just to have the poses folder. I am unfamiliar with Kitti data structure, so I appreciate the help. All is working as expected, thanks!

closed 2022-10-11 by espackman