PRBonn/kiss-icp #12: Which seq is the preview demo video used?

Kin-Zhang, 2022-10-11:
https://user-images.githubusercontent.com/38326482/189950820-030fd9e4-406b-4d14-8171-43b134344223.mp4

Hi, I'm wondering which seq is the preview demo video used? Thanks in advance!

nachovizzo, 2022-10-12:
1. Sequence 08
2. As explained in the paper, we do not implement loop-closure
3. I can't understand this question, but if the rotation is huge, there is nothing KISS-ICP can do.
4. We did ablation studies internally, and indeed there is never a real difference. We even tried with iKDtrees. The acceleration data structure is just used to speed up NN-seraches, nothing else. 
5. We don't have an experiment for supporting this, but we've seen the system struggle with 16-beams datasets.

closed 2022-10-14 by nachovizzo

Kin-Zhang, 2022-12-06:
Thanks for the reply.

> As explained in the paper, we do not implement loop-closure

Without this one, I think maybe it may fail a lots of suitation. Did you also do some experiments on the how long will make the odom not so accurate and how to reduce this effect?

---
Sorry in advance because the following may be considered an offense. Actually not, just my confusion.

I rechecked the paper, and after reading it (but not so detailed sorry), I was still confused about **why** the kiss-icp could **make the system simple**. It's pretty much similar core-method to years ago papers like icp and ndt

for the discussion of the parameters. Maybe [A-LOAM only has two parameters](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/devel/launch/aloam_velodyne_VLP_16.launch) in their launch file to change:
```bash
    <param name="mapping_line_resolution" type="double" value="0.2"/>
    <param name="mapping_plane_resolution" type="double" value="0.4"/>
```

also [autoware.ai 1.13](https://github.com/autowarefoundation/autoware_ai_perception/tree/master/lidar_localizer) who open-sourced at 2019 also involved with ndt methods and their parameters could also not so many for 7/8 if you set same resolution all the time.

And there are also many methods (of course you mentioned some in the paper) like with only one LiDAR could also get the odom/pose for point cloud: [hdl_graph_slam](https://github.com/koide3/hdl_graph_slam), [BLAM](https://github.com/erik-nelson/blam), [RAL2022 VoxelMap](https://github.com/hku-mars/VoxelMap)

Through all these comparisons (all methods I mentioned above could run only on one LiDAR without IMU, camera etc), maybe you could explain more on how your method could be simple without tuning so many params, and also which part makes the result better than others or above methods.

And of course, and still, you did really great thing to make the icp method open for everyone through python pip install. I also learned a lot from your other works.