PRBonn/kiss-icp #19: Error after visualization finishes.

espackman, 2022-11-02:
Hi all,

I am using Kiss ICP on the Kitti dataset sequence 07. 
After the visualization finishes (looks awesome, by the way), I get the following error:
"'cat' is not recognized as an internal or external command, operable program or batch file".

The results section appears to output as expected. 
Is this error a cause for concern? 

Thank you

nachovizzo, 2022-11-04:
Wow, I didn't see this one coming. To write the results to the file disk, I use a very inefficient and ugly technique:

```python
results_fn = f"{self.results_dir}/result_metrics.log"
with open(results_fn, "w") as file:
    stdout = sys.stdout
    sys.stdout = file  # Change the standard output to the file we created.
    print(f"Results for {self._dataset.__class__.__name__}:{self._dataset.sequence_id}")
    print(f"AVG translational error:{avg_trans_error:.3f}")
    print(f"AVG rotational    error:{avg_rot_error:.3f}")
    print(f"ATE translation     [m]:{ate_trans:.3f}")
    print(f"ATE rotation      [rad]:{ate_rot:.3f}")
    print(f"AVG FPS           [fps]:{fps}")
    sys.stdout = stdout
os.system(f"cat {results_fn}")
```

Basically, I write to a file and then `cat` this file to the standard output. The problem you have is that you do not have `cat`, `sudo apt install coreutils` should solve the problem. In any case, the file was most-likely successfully created, and you can find it in the results folder of your current working directory.

espackman, 2022-11-04:
Thanks! I will close out this ticket.

closed 2022-11-04 by espackman