PRBonn/kiss-icp #20: When will the full source code be released?

espackman, 2022-11-04:
Hi there,

I really like what you guys have done. I would like to take a look at the source code, and saw the comment on the main page that the code will be released once the review process of the paper is complete.

Is there an estimation of when this will be completed? Thank you!

nachovizzo, 2022-11-07:
Hello @espackman, we sadly regret to inform that the paper did not get accepted, and now we are in the process of revising it and resubmit. This will take another 1, 1.5 months. I'm sorry we are making a lot of people wait for the code, but we can't control who is reviewing the paper or how.

Make sure to "watch"(click on the icon) the repo, you will get notified anytime there is a new release with the full implementation

closed 2022-11-07 by nachovizzo

espackman, 2022-11-07:
Hi @nachovizzo,
I'm sorry to hear that. You guys have made some incredible work, so I'm sure its just a matter of time before it is officially accepted. I'm more than happy to wait patiently; thank you for the update!

Best of luck.

nachovizzo, 2023-01-11:
@espackman Source code is already out now! [v0.0.13](https://github.com/PRBonn/kiss-icp/tree/v0.0.13)

michalpelka, 2023-01-11:
Great! Codebase looks very simple and clean. I will test it soon. Thank you for the release.