PRBonn/kiss-icp #21: No results

GabiShahmayster, 2022-11-13:
Hi,

First of all, think you very much for providing the library, it looks very promising.
I am trying to run a scenario where a folder contains several point-cloud files, in a PLY format. These files contain de-skewed Velodyne data.
So far, running the pipeline results in a rapidly progressing progress bar and no results other than a clone of the config file...
Is there some way to understand/log any issues which might cause such behavior?

Thank you in advance,
Gabi

nachovizzo, 2022-11-14:
Hello @GabiShahmayster, is hard to say without looking at the data. Did you try to `--visualize` to see what the system is doing? Do you know if the pointclouds are in other reference frame, perhaps?

GabiShahmayster, 2022-11-14:
@nachovizzo Thanks for the reply.

Unfortunately, I'm having an issue with visualization... getting the following message:
OSError: libX11.so.6: cannot open shared object file: No such file or directory
It's probably related to the fact that I'm using a very simple docker to test the library, a pretty bare-bone one which comes with python 3.8 (followed by pip installing kiss_icp + open3d)

The point cloud is captured using a Velodyne installed on a vehicle, the reference frame is forward-left-up and the data is de-skewed using an independent ego-motion estimation algorithm. We know that the data is valid and properly de-skewed...

GabiShahmayster, 2022-11-14:
@nachovizzo 
Thanks! That worked flawlessly.
I can now see the point-cloud and it looks great. 
The only problem is that the pose outputs file (which I understand to be in Kitti format) is nowhere to be found...
Gabi

nachovizzo, 2022-11-14:
@GabiShahmayster I'm happy to hear that!

The output folder is usually on the current working directory, although you can download the config file and change the location to wherever you want:
https://github.com/PRBonn/kiss-icp/blob/a3e7444f1e33ca901b48375f621336440fd2fb98/config.yaml#L1

And yes, it's not yet well documented, but the poses are in kitti format, so `evo_traj` and all kitti-friendly tools can operate over this file

GabiShahmayster, 2022-11-14:
@nachovizzo 
Thank you.
The poses output file is still not there.
I tried using a config file (the same one which was automatically written to the "latest" output folder), which resulted in the following error:

```
ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object/new:easydict.EasyDict'
  in "results/latest/config.yml", line 1, column 21

```

The config file itself is:
`adaptive_threshold: !!python/object/new:easydict.EasyDict
  dictitems:
    initial_threshold: 2.0
    min_motion_th: 0.1
  state:
    initial_threshold: 2.0
    min_motion_th: 0.1
data: !!python/object/new:easydict.EasyDict
  dictitems:
    lidar_frequency: 10.0
    max_range: 100.0
    min_range: 5.0
    preprocess: true
  state:
    lidar_frequency: 10.0
    max_range: 100.0
    min_range: 5.0
    preprocess: true
mapping: !!python/object/new:easydict.EasyDict
  dictitems:
    max_points_per_voxel: 20
  state:
    max_points_per_voxel: 20
out_dir: results
use_cache: true
`

nachovizzo, 2022-11-16:
@GabiShahmayster This sounds like something unrelated to kiss-icp, maybe is because you have no write permisions? 

Anyway, I'm happy it worked out with rosbags (FYI, the rosbag dataloader is SLOW, you can get a 5x improvement on runtime if you are not using rosbags)

closed 2022-11-16 by nachovizzo

nachovizzo, 2022-11-16:
You are right, could you send me privately a small dataset? This why I can debug it myself and provide a quick solution ;)

nachovizzo, 2022-11-16:
Thanks! So, it's working, but the problem is that we still did not release the full documentation for the project. The issue is that the "sequence name" is being taken by the folder name. Actually in the data you sent me, the results are there, but sadly the name is `..txt`.

## Solution

```sh
mkdir -p gabi_pcd_example
mv *.pcd ./gabi_pcd_example/
kiss_icp_pipeline --visualize  /gabi_pcd_example/
```

Enjoy the results (it looks to be working!) and then, use any tool you like to plot the results:

```sh
evo_traj kitti results/latest/gabi_pcd_example.txt -p
```
[image]

## Recommendations

This data looks quite challenging, to make your life easier, you could create a custom dataloader, and for example flip the Z axis of the lidar

## Video

This is the out-of-the-box results I obtain (you can get better with some tunning, but not needed)

https://user-images.githubusercontent.com/21349875/202154522-e4cd4ca5-8a37-457f-95fa-d245244e44d0.mp4

GabiShahmayster, 2022-11-16:
Thank you very much, problem solved.

On a personal note, it's very rate to find a complex algorithm which works so well out of the box, with zero tweaking. Amazing job.