PRBonn/kiss-icp #22: MulRan Dataset loader

fatemechangizian, 2022-12-02:
Hello, 
First thanks for the great work.

I have a general question about the pre-defined loaders, you have for some benchmark samples like kitti, Mulran and etc.
I would like to done some test on Mulran sample which consists of a few sequences as ["kaist/KAIST01", "kaist/KAIST02", "kaist/KAIST03", "dcc/DCC01", "dcc/DCC02", "dcc/DCC03", "riverside/Riverside01", "riverside/Riverside02", "riverside/Riverside03", "sejong/Sejong01", "sejong/Sejong02", "sejong/Sejong03"]
But I couldn't find a link for downloading those samples on the Mulran webpage. Is it possible if I can know how can I get access these samples? (I can see the explanation on the website but no link to download)

Thanks and looking forward to your response.

nachovizzo, 2022-12-02:
Hello, you need to go to the official [website](https://sites.google.com/view/mulran-pr/download) and request for access. You will receive an email with the Google Drive links, I tried to maintain the same folder structure they had there, but since it's a GDrive link this can change overtime.

## Expected dataset layout

```
MulRan
├── dcc
│   ├── DCC01
│   │   ├── data_stamp.csv
│   │   ├── global_pose.csv
│   │   ├── gps.csv
│   │   ├── navtech_top_stamp.csv
│   │   ├── Ouster
│   │   │   ├── 1564718964199537916.bin
│   │   │   ├── 1564718964300400595.bin
│   │   │   ├── ...
│   │   ├── ouster_front_stamp.csv
│   │   └── xsens_imu.csv
│   ├── DCC02
│   └── DCC03
├── kaist
│   ├── KAIST01
│   ├── KAIST02
│   └── KAIST03
├── riverside
│   ├── Riverside01
│   ├── Riverside02
│   └── Riverside03
└── sejong
    ├── Sejong01
    ├── Sejong02
    └── Sejong03
```

fatemechangizian, 2022-12-02:
Oh okay, I'm really surprised I haven't seen that before. :|
Thank you for your really quick response.