PRBonn/kiss-icp #24: Full release date?

espackman, 2022-12-23:
Hi team,

I hope the holidays and the upcoming new year find you all well!
From previous posts, I understand that the full release of KISS-ICP will occur once the paper is accepted, but I haven't seen any recent news.
Are there any updates or an expected timeline for acceptance?

Thank you and wishing the best,
Everett

nachovizzo, 2023-01-02:
Hello @espackman, sorry for the delay on the response. We are delighted to announce that the paper has been accepted for publication and therefore, we are releasing the source code within the next days! I will keep this issue open, so I can notify you first once it's out there ;)

espackman, 2023-01-06:
@nachovizzo That's great to hear! I will keep an eye out for the notification :)

nachovizzo, 2023-01-11:
Already out there! [v0.0.13](https://github.com/PRBonn/kiss-icp/tree/v0.0.13)

closed 2023-01-11 by nachovizzo