PRBonn/kiss-icp #27: Pure cpp example?

JaySlamer, 2023-01-12:
Thank you for your great work.
I don't know python so I feel a bit lost. 
I would like to know how to load two point clouds and output the transform in pure cpp using kiss-icp.
I took a look at registration.cpp. it seems it only has one parameter: th. If I use registration.cpp, can I reap the benefit of kiss-icp or am I missing a lot stuff?
Would you be so kind to construct of a pure cpp example registering two point clouds using kiss-icp.

nachovizzo, 2023-01-12:
Hello @JaySlamer, yes, this is a problem indeed. Now we have a mixture between C++/Python. But from next week on, I will start working in a proper ROS wrapper that does not enforce python as dependency. Will this work in your case? Or do you think it is still valuable to have a C++ CLI?

I will keep this issue open as a reminder to push for the ROS wrapper

PS: The "big" missing part on the C++ is only the adaptive threshold estimation, but if you use a fixed threshold(that also works pretty well) then all the ingredients are already on the C++ side.

huawei-sai, 2023-01-12:
@nachovizzo Thank you for this excellent work. 

Having a ROS wrapper with only C++ implementation would make easy to integrate with existing pose-graph back ends (https://github.com/gisbi-kim/SC-A-LOAM/blob/main/src/laserPosegraphOptimization.cpp). I would vote for this and will be waiting to test it myself.

The main reason to use ROS wrapper is to run the SLAM systems on-device easily as most of us use ROS for communicating the data throughout the robot, both front-end and other perception modules.

JaySlamer, 2023-01-12:
> Hello @JaySlamer, yes, this is a problem indeed. Now we have a mixture between C++/Python. But from next week on, I will start working in a proper ROS wrapper that does not enforce python as [...]

A ROS wrapper would be great. Thank you.

chengwei0427, 2023-01-13:
It's a good idea for a ROS wrapper with only C++ implementation.

I'll check and update.

Thank you.

nachovizzo, 2023-01-19:
@JaySlamer Not yet there, but [here](https://github.com/PRBonn/kiss-icp/blob/main/src/cpp/kiss_icp_ros/KissICP.cpp) is a draft on how to use the system only with C++

PR #49 (2023-01-24): Manual merge, add ROS1 support
Add preliminary ROS1 support

Closes #27
closed 2023-01-24 by nachovizzo