PRBonn/kiss-icp #28: Installation unsuccessful

jeychandar, 2023-01-12:
@stachnis Hi I am new to this field. I studied your work it was amazing. Interested to know how this works. As per your readme file I have installed everything but still shows **kiss_icp_pipeline:Command not found**. Am I missing any dependencies?.Can you clarify me. Thanks in advance

nachovizzo, 2023-01-12:
Hello @jeychandar there is no need to tag anyone to ask questions, just write it down here ;)

It's hard to guess why is not working in your case, since you didn't provide any information, OS? How did you install it?, etc. Please clarify a bit further your setup, so we can help you with the installation. ;)

jeychandar, 2023-01-12:
OS: Ubuntu 20.04.5 LTS
I am using this version of OS for LIO-SAM.
[image]

jeychandar, 2023-01-12:
> 

Thanks It worked. Will explore more now :)

closed 2023-01-12 by jeychandar

h-rover, 2023-01-12:
I am having the same issue. I can import it once inside the python but cannot find the package from the command line.

`rover@rover:~$ python
Python 3.6.9 (default, Nov 25 2022, 14:10:45) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import kiss_icp
>>> 
[1]+  Stopped                 python
rover@rover:~$ kiss_icp_pipeline --help
kiss_icp_pipeline: command not found
hamza@hamza:~$ 
`
My bashrc has the exported paths as follows,
`export PATH="/home/rover/.local/bin:$PATH"
export PYTHONPATH="${PYTHONPATH}:/home/rover/.local/lib/python3.6"
export PYTHONPATH="${PYTHONPATH}:/home/rover/.local/lib/python3.6/site-packages"`

Any help will be appreciated.

h-rover, 2023-01-12:
I am having the same issue. I can import it once inside the python but cannot find the package from the command line.

```
rover@rover:~$ python
Python 3.6.9 (default, Nov 25 2022, 14:10:45) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import kiss_icp
>>> 
[1]+  Stopped                 python
rover@rover:~$ kiss_icp_pipeline --help
kiss_icp_pipeline: command not found
rover@rover:~$ 
``` 

I am having the same issue. I can import it once inside the python but cannot find the package from the command line.

My bashrc has the exported paths as follows,

```
`export PATH="/home/rover/.local/bin:$PATH"
export PYTHONPATH="${PYTHONPATH}:/home/rover/.local/lib/python3.6"
export PYTHONPATH="${PYTHONPATH}:/home/rover/.local/lib/python3.6/site-packages"`
```
Any help will be appreciated.

nachovizzo, 2023-01-12:
Why did you manually typed your home path? Can you paste the output of $echo $PATH? If you can import it cut can't access the binary is a miss configuration on your system most likely

h-rover, 2023-01-13:
Thanks for the quick response. Exporting the python path in the bash script fixed it!