PRBonn/kiss-icp #29: How to run kitti Dataset

jeychandar, 2023-01-12:
> Great! We will update the requirements once the final release is out there

Hi I am working with exisited ros bag for kitti dataset provided by https://github.com/TixiaoShan/LIO-SAM. I am trying to play with ros2 bag distro(foxy). how to play the recorded bag file. Ros2 version supports your repo?. Can you please share the command line to play the recorded bag.

_Originally posted by @jeychandar in https://github.com/PRBonn/kiss-icp/issues/6#issuecomment-1380070084_

nachovizzo, 2023-01-12:
@jeychandar so, this is NOT a  ROS1 bagfile, I just downloaded and run:
```sh
$ kiss_icp_pipeline 2011_09_30_drive_0028.bag --topic /points_raw --visualize
``` 

It's working:
[image]

But I'm not sure why you want to use that particular bagfile, there is no `timestamp` information, which means that you won't be able to "deskew" the pointclouds. I recommend using the KITTI dataloader instead.

These are the poses obtained by KISS-ICP, **without** deskewing:

[image]
[2011_09_30_drive_0028_poses_kitti.txt](https://github.com/PRBonn/kiss-icp/files/10400773/2011_09_30_drive_0028_poses_kitti.txt)

**At this point, PLEASE open a new issue since this is not at all related with ROS2 support**

jeychandar, 2023-01-12:
> @jeychandar so, this is NOT a ROS1 bagfile, I just downloaded and run:
> 
> ```shell
> $ kiss_icp_pipeline 2011_09_30_drive_0028.bag --topic /points_raw --visualize
> ```
> 
> It's working: [image]
> 
> But I'm not sure why you want to use that particular bagfile, there is no `timestamp` information, which means that you won't be able to "deskew" the pointclouds. I recommend using the KITTI dataloader instead.
> 
> These are the poses obtained by KISS-ICP, **without** deskewing:
> 
> [image] [2011_09_30_drive_0028_poses_kitti.txt](https://github.com/PRBonn/kiss-icp/files/10400773/2011_09_30_drive_0028_poses_kitti.txt)

I am happy that it is working and thanks for sharing the output.The problem that I am facing is I am not able to visualize.I didn't noticed that this particular bag doesn't have timestamp sorry for that.I used kitti dataloder also but still showing the same error sir.

jeychandar, 2023-01-12:
kiss_icp_pipeline /home/praveen/Downloads/2011_09_30_drive_0028.bag --topic /points_raw --visualize
I worked the following command and I am not able to visualize the output like you. It shows the error like
[image]

nachovizzo, 2023-01-12:
This looks unrelated to kiss-icp. The error says **ImportError: cannot import name 'Log' from 'rosgraph_msgs.msg' **

You can google and check realted issues like https://github.com/ros/ros_comm_msgs/issues/13

jeychandar, 2023-01-12:
Yes Sir I have checked. But this error came up with **kiss_icp/ datasets/rosbag.py** line 57. The bag automatically closes.

nachovizzo, 2023-01-12:
Yes, but the import error is outside kiss-icp. Please try this in a dockerize environment, so I can reproduce the error, otherwise there is nothing I can do from my side.

jeychandar, 2023-01-13:
I am able to run successfully. It creates results folder inside that it generated npy format and text format and config file. Is there any command line to view the poses it generated.

nachovizzo, 2023-01-13:
Great news!

Then you can use `evo_traj` t plot the trajectory, `pip install evo` and then you can run something like this
```sh
evo_traj kitti results/latest/*.txt -p --plot_mod=xy
```

closed 2023-01-13 by nachovizzo

jeychandar, 2023-01-13:
Great it works.Actually I didn't use docker environment.The file which is created using ros1bag and I tried to run in ros2. So went back to his studies LIO-SAM and got some ideas about the dynamic bridge and ran your command.It works perfectly.Thank you for your guidance Sir :). 
[image]