PRBonn/kiss-icp #34: Visualisation range, or internal map range?

rlabs-oss, 2023-01-12:
open3d viewport needs to be larger, or the internal map size. the pointcloud displayed during visualization will populate to a certain range (not the lidar range) which causes mapping failure. As the odometry / local frame reaches the edge of this region the map will collapse. - there be dragons.

nachovizzo, 2023-01-13:
It's not clear what is the action needed from our side.

Please also check the help message of the visualizer, there are 2 modes, 1 it's with the global point of view and the second one is with an egocentric visualization. The local map used for registration moves along with the pose of the vehicle, meaning that there is **no global map at all**. All the details of our system can be found in the paper ;)

```sh
RegistrationVisualizer initialized. Press:
        [SPACE] to pause/start
          [ESC] to exit
            [N] to step
            [R] to toggle on/off the raw frame
            [F] to toggle on/off the input cloud to the pipeline
            [K] to toggle on/off the subsbampled frame
            [M] to toggle on/off the local map
            [V] to toggle ego/global viewpoint
            [T] to toggle the trajectory view(only available in global view)
            [C] to center the viewpoint
            [W] to toggle a white background
            [B] to toggle a black background
```

closed 2023-01-13 by rlabs-oss

rlabs-oss, 2023-01-13:
If the input cloud is already in global frame (from faster_lio) - kiss-icp will still register the clouds, but will eventually fail as the range of the input cloud (since those points are already translated) is exceeded.  - This isn't a typical scenario but it was the test data I had at the time. :/