PRBonn/kiss-icp #35: Robust kernel formula (code vs paper)

pewpeo, 2023-01-13:
Hi! At first, thank you for your great work!

I am really interested in this topic since I was working a long time with lidar odometry algorithms and I think I share your excitement regarding simplicity and independence of different sensors and platforms.

I was just reading the paper and digging through the implementation. Doing this, one question came up regarding the following line:

https://github.com/PRBonn/kiss-icp/blob/743fa5a5725fe5d7576551c5f7c1a00b70ae63e6/src/kiss_icp/cpp/kiss_icp/Registration.cpp#L105

Can you explain how this coincides with formula 13 from the paper (the robust kernel):

$$ p(e) = \frac{e^2/2}{\kappa_t + e^2} ~~, \text{with} ~~ \kappa_t = \sigma_t/3 $$

I already found that `th` equals `sigma/3`, so this would be $\kappa_t$ in the paper and `residual2` would be $e$. But I don't see that this equals this formula. What I see is:

$$ p(e) = \frac{\kappa^2}{(\kappa + e)^2} $$

nachovizzo, 2023-01-13:
Hello, @pewpeo thanks for checking out the code in this detail. Indeed, the formula you have is correct, and it's designed to be like this. This is the weight used in the optimization, not the same formula of the robust kernel.

I would recommend checking a tutorial I wrote some time ago for [Open3D](http://www.open3d.org/docs/release/tutorial/pipelines/robust_kernels.html), which is based on [this paper](https://arxiv.org/pdf/1810.01474.pdf):

[image]

pewpeo, 2023-01-14:
Aaah... that was the connection I missed. Thank you for your answer and the links! :)

closed 2023-01-14 by nachovizzo