PRBonn/kiss-icp #36: Difference between paper with implementation?

mertyavuz41, 2023-01-15:
Hi,

First of all, I would like to thanks to great studies. When I examine the code I observed that difference between mentioned paper deskewing formula and implemented deskewing method. 

Formula has exp process but I could not see exp process on the code. 
Paper:
[image]

Code:
https://github.com/PRBonn/kiss-icp/blob/a31f5396c6153f12e9011c9ae6d061597d2f45b3/src/cpp/kiss_icp/Deskew.cpp#L58-L63

Am i missing something?

nachovizzo, 2023-01-15:
If you read 1 line below that formula, you will find the answer to your question: 
[image]

It's not an "exponential" function, but an exponential map. I'd recommend checking out some lie algebra theory ;)

closed 2023-01-15 by nachovizzo

mertyavuz41, 2023-01-15:
Thanks for your informative feedback 🙏