PRBonn/kiss-icp #37: Is kiss-ICP work with 2D LIDARs ?

LotfiZ, 2023-01-15:
Hi,
First, thank you for this amazing work, its really interesting and im looking forward to work with it ! 
My question is quite straight forward : is it possible to have the same result if we want to work with 2D LIDARs ? or this approach is exclusively used for 3D LIDARs ? 

Best regards

nachovizzo, 2023-01-16:
Hi, thanks for the interest in the project

 In principle, it should work. KISS-ICP can process anything that can be expressed as 3D Points, so even if you add a `z=0` channel to your data, the system should do something meaningful. We recently tried on 2D Radar data and was working flawlessly. 

Do you have any particular dataset where we could try it?

LotfiZ, 2023-01-17:
Thank you for the quick answer. Unfortunately, i don't have a 2D Lidar dataset. However i can generate it from a 2D safety laser (SICK), i'll provide it as soon as possible.

Thanks again

nachovizzo, 2023-01-17:
Ok, feel free to drop a link here whenever you have some data :)

closed 2023-01-17 by nachovizzo

LotfiZ, 2023-01-20:
Hi @nachovizzo,

As promised, i recorded two ROS1 bag files of `sensor_msgs/PointCloud2` messages from two different AGVs : 

1.  [https://drive.google.com/file/d/1Ki5KhhwZDcTuixTPZ4f8ukQOqfbw5IKq/view?usp=share_link](url) Recording of data from a differential AGV that has 2 SICK MicroScan Safety Lasers, i merged the data to have one single source of laser points  (thanks to [https://github.com/iralabdisco/ira_laser_tools](url) ) 
2.  https://drive.google.com/file/d/17tieKtJIKy3y2zwV_wU_ThQDLp6SKlkh/view?usp=share_link Recording of data from a forklift that has 3 SICK MicroScan Safety Lasers, i merged the points same as the first AGV.

If there is anything that is missing or data mismatch with kiss-ICP. Please feel free to ask me and i can provide better data.

Thanks

nachovizzo, 2023-01-23:
@LotfiZ Hello, the first link I can't access and the second asked me for a password :D

LotfiZ, 2023-01-23:
Sorry for the inconvenience...i did not well setup the access of the links ! I think that now its fixed.

nachovizzo, 2023-01-25:
@LotfiZ Thanks for sharing the data, the first bag I couldn't figure out how to deal with it. But the second one I just tried it with the pipeline and the results looks good!

Just `pip install kiss-icp`, to obtain better results I've changed the `min_range` and `max_range` in the config.yaml to match more the data type:

```sh
$ kiss_icp_pipeline ~/Desktop/FL_2D_LASERS.bag --topic /merged_cloud --visualize --config ./config.yaml
````

config.yaml
```yaml
out_dir: "results"

data:
  deskew: False
  frame_rate: 10.0 # <- optional only valid if deskew==True
  preprocess: True
  max_range: 20.0 # can be also changed in the CLI
  min_range: 2.0

mapping:
  max_points_per_voxel: 20

adaptive_threshold:
  initial_threshold: 2.0
  min_motion_th: 0.1
```

These are the results I've obtained

## Run
https://user-images.githubusercontent.com/21349875/214510116-194f52d0-abd3-45e8-a6e7-9133dfc010ce.mp4

## Poses
[image]

So I guess that it's actually working out of the box!

LotfiZ, 2023-01-25:
Amazing ! what a great work you did. Congrats for all your team ! If you want/need more 2D lasers data or test on real systems, feel free to ask !