PRBonn/kiss-icp #39: dataloader for TUM RGB-D SLAM Datasets

lsorgi, 2023-01-16:
Which dataloader can be used to process data from TUM RGB-D SLAM Dataset and Benchmark? (https://vision.in.tum.de/data/datasets/rgbd-dataset)

If no dataloader is available, where can I find the signature of the dataloader class to implement ?

thanks

nachovizzo, 2023-01-16:
Hello, thanks for the interest in our project.

I just made a draft of the dataloader, but there is still some details to fix. Could you please give it a try and provide some feedback to see if it's working?

## How to try it?
```
git clone  https://github.com/PRBonn/kiss-icp.git
cd kiss-icp
git checkout nacho/add_tum_dataloader
```

After that, you can ran `pip install --verbose .` to install this branch

## How to run?

```sh
kiss_icp_pipeline --dataloader tum $DATASETS/tum/rgbd_dataset_freiburg1_xyz --visualize
```

## Expected output

[image]

The evaluation is still wrong but we are working on finishing the draft of the dataloader.

lsorgi, 2023-01-16:
hello Ignazio, many thanks for your prompt answer. 
and thanks for making all this opensource. great work :-)

lsorgi, 2023-01-16:
feedback about installation on Windows laptop:

the command 
`pip install --verbose .`
fails with the following error
```
ERROR: Failed building wheel for kiss-icp
  Failed to build kiss-icp
  ERROR: Could not build wheels for kiss-icp, which is required to install pyproject.toml-based projects
```
 
the command 
`pip install -e ".[all]"`
completes successfully

lsorgi, 2023-01-16:
I tried to patch the pyproject.toml file and I get the same error.  (trying now to build on Ubuntu 18.04)
Here is the full log from Windows.

[setup_log.txt](https://github.com/PRBonn/kiss-icp/files/10427617/setup_log.txt)

nachovizzo, 2023-01-16:
Then try with `pip install -U cmake ninja`

nachovizzo, 2023-01-16:
pip version? Are you using conda or any other animal alike? The problem is that the build system is not finding ninja on the path.

lsorgi, 2023-01-16:
yes using similar animal, miniconda

nachovizzo, 2023-01-16:
Ah ok, then I can't help with that, since I'm allergic to those animals :P they never really worked for me :D.

kiss-icp does not have many dependencies, so there is actually no big gain on isolating the whole thing ;)

lsorgi, 2023-01-16:
I'll do a few more attempts within the conda env.

- I think I managed to install correctly ninja  (https://anaconda.org/conda-forge/ninja)
- the build error is different now, maybe you see somethin
[Ubuntu1804_setup_log.txt](https://github.com/PRBonn/kiss-icp/files/10427844/Ubuntu1804_setup_log.txt)
g?

nachovizzo, 2023-01-17:
I see the same error:
```
'/tmp/pip-build-env-d5sl4e3f/overlay/bin/ninja' '--version'

    failed with:

     No such file or directory
```

Now you know why I don't like conda :p. Too complicated to manage just some python dependcies 😇

Have you tried outside conda?

lsorgi, 2023-01-17:
issue fixed! 
g++-9 is needed, nothing to do with anaconda. you may want to include this in the readme.
what I did is 
- install g++-9
- update symbolic link g++ -> g++-9 executable
and the build was successful

nachovizzo, 2023-01-17:
Great! It's actually already on the [Readme](https://github.com/PRBonn/kiss-icp#install-developer-mode). But I don't want to enforce any particular version, since it depends on your OS

lsorgi, 2023-01-23:
I am trying to process the TUM dataset in a simple loop but the final map does not match the one showed above. 
Below is pseudo code & the KISS configuration. Anything wrong in the way I am using the library?

    
	versor_map = ...  # this is HW x 3 matrix containing the 3d unit norm vector corresponding to each pixel 
	
	for filename in depth_file_list:
              logging.info(f'Processing data file ({filename})')

              # extract timestamp from filename
             timestamp, _ = os.path.splitext(os.path.basename(filename))
             timestamp = float(timestamp)
        
             # load depth map
             depth_map = cv2.imread(filename, cv2.IMREAD_ANYDEPTH)

            # convert to 3D data        
            depth_map = depth_map.flatten() * 1e-2  # convert to [cm]
            flags = depth_map > 0  # exclude invalid depth (depth == 0)
            pts_3d = versor_map[flags, :] * np.tile(depth_map[flags, np.newaxis], reps=(1, 3))

            # run SLAM
            odometry.register_frame(frame=pts_3d, timestamps=timestamp)

          KISS-ICP Configuration: 
          {
              'adaptive_threshold': 
              {
                   'fixed_threshold': 0.3,
                   'initial_threshold': 1.0,
                   'min_motion_th': 0.01
              },
              'data': 
               {
                     'correct_scan': True,
                     'lidar_frequency': 10.0,
                     'max_range': 100.0,
                     'min_range': 0.0,
                    'preprocess': True
              },
             'mapping': 
             {
                  'max_points_per_voxel': 20, 
                  'voxel_size': 1.0
             },
             'out_dir': 'results'
     }

nachovizzo, 2023-01-23:
@lsorgi All the development for the TUM dataset is happening at #45 . I suggest you stay tunned for when we merge this into the main branch so you can give it a try

nachovizzo, 2023-01-23:
@lsorgi  we just merged #45 please let us know if everything is working on your side!

closed 2023-01-23 by nachovizzo