PRBonn/kiss-icp #43: online SLAM

lsorgi, 2023-01-17:
Hello, 

is there a way to use the library for online data processing? (see pseudocode below)

`
...
while True:
  pts_3d = acquire_data_from_depth_sensor()
  pose = kiss_icp(pts_3d)
`

nachovizzo, 2023-01-17:
That wouldn't be a SLAM application, but leaving that detail aside 😉 we are currently working on a ROS wrapper that can be used for this same application.

I'll try to notify you when it's ready but you can enable notification in this repo (press watch) in case I forget to do so!

closed 2023-01-17 by nachovizzo

lsorgi, 2023-01-17:
oki! 
sure, I didn't include the 'mapping' part of SLAM because I am mostly interested on camera tracking. 
thanks for the hint, I'll stay connected to see the evolution of the repo.

nachovizzo, 2023-01-25:
> That wouldn't be a SLAM application, but leaving that detail aside 😉 we are currently working on a ROS wrapper that can be used for this same application.
> 
> I'll try to notify you when it's ready but you can enable notification in this repo (press watch) in case I forget to do so!
@lsorgi:
ROS wrapper is already available on the main branch 😉