PRBonn/kiss-icp #51: ROS2 Datasets Wanted

nachovizzo, 2023-01-26:
I have a preliminary version of KISS-ICP working in a ROS2 environment, but I'm having trouble to find some mobile robots public datasets out there to try out the system.

So far, I could only try the bagfile (rather small) from #6. But I would like to test it on more outdoor scenarios. If you have some data you would like to try, please send it to us! Link, GDrive, etc.
[image]

[image]

ghfue, 2023-01-26:
With the limited ROS2 datasets available, I would just convert the ROS1 datasets to ROS2, probably using something like the [rosbags](https://gitlab.com/ternaris/rosbags) tool. Most of the ROS1 datasets use standard ros messages, so it should be a straightforward conversion.

fratopa, 2023-01-26:
You could try [this dataset](https://arxiv.org/abs/2207.06815). It has IMU, Lidar and stereo camera data. I converted the bag to ros2 using [this tool](https://gitlab.com/ternaris/rosbags) and the visual output worked flawlessly with ros2. I still need to check how the pointcloud gets converted.

asaba96, 2023-01-26:
@nachovizzo I have quite a few datasets that I would love to try this on! The data is from our autonomous racecar (checkout the [Indy Autonomous Challenge](https://www.indyautonomouschallenge.com/))

The datasets I have are fairly large and are from multiple LiDARs (3 luminar hydras to be specific), configured to provide a 360 degree FOV. I have the ability to provide bags that have all 3 LiDAR streams on one topic, or bags that have each stream on its own topic. The accumulated cloud is cropped (40m by 200m) but I have the ability to recreate the original full-size cloud.

There are also 2 Dual-Antenna RTK-GNSS+IMU systems on board, so "ground truth" is available.

Happy to collaborate/help!

doisyg, 2023-01-26:
We also have a couple of data sets that we could share. Indoor warehouse environment with 2 360x90 FoV Lidars + IMU.

nachovizzo, 2023-01-27:
> @nachovizzo I have quite a few datasets that I would love to try this on! The data is from our autonomous racecar (checkout the [Indy Autonomous [...]

I'd love to try it, where can I find the dataset?

ga58lar, 2023-02-02:
@nachovizzo I am a Ph.D. Student from the Institute of Automotive Technology from the TU Munich and we also have quite a substantial amount of ros2 datasets from the Indy Autonomous Challenge, but we also have data from our new autonomous research vehicle, which we are happy to share with you!  
  
We also already tested your algorithm with a dataset of our campus and the results are really spectacular. Thank you for the great work!  
[image]
credits to @markuspi

michalpelka, 2023-02-17:
@nachovizzo  I can provide a simulated environment with LiDAR, with https://github.com/RobotecAI/o3de-rgl-gem , here is a sample : https://drive.google.com/file/d/1cuGnpyQEyDsKlYJIdAE5D-YxvGTjOCdj/view?usp=sharing
Also, I have Livox Mid360 at my disposal, so I can try to set up data capture with ROS2 and collect some small datasets for you handheld.

nachovizzo, 2023-02-20:
@michalpelka, @ga58lar Thanks! Yes please share as much data as you guys have, so I can make sure it is more or less working on multiple scenarios ;) If you have no problems you can paste the links here so other people can also use it.

I guess soon I will merge the ROS2 wrapper into the main public branch, so it will also be available for you guys.

#71 is the first step towards this goal ;)

closed 2023-02-23 by nachovizzo

nachovizzo, 2023-02-23:
@michalpelka I've tested the ROS2 wrapper on your data and looks to be working on my side. Thanks a lot for sharing!