PRBonn/kiss-icp #53: catkin build : undefined reference to pthread

martinlucan, 2023-01-27:
Hi there, 
 I face the following error during catkin build. Has anyone come across the same?

```
kiss_icp/src/cpp/kiss_icp/tbb/lib/libtbb.a(governor.cpp.o): undefined reference to symbol 'pthread_key_delete@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
```

Thank you in advance,
Martin

nachovizzo, 2023-01-27:
@martinlucan can you specify a bit more about your system, etc? In any case, one option would be to install the TBB library locally. Just do `sudo apt install libtbb-dev` and try to build

martinlucan, 2023-01-27:
@nachovizzo I run ubuntu 18.04 on x86 arch. with ROS Melodic. Unfortunately, libtbb-dev is already installed.

nachovizzo, 2023-01-27:
I'm sorry, but I can't help in this case, the Operating system is 3 months away from being deprecated ("April 30, 2023") I'd suggest you upgrade to something more modern

closed 2023-01-27 by nachovizzo

nachovizzo, 2023-01-27:
@martinlucan I anyways found a workaround, but I still would recommend to upgrade the system. 

I tried on a docker enviroment: ` osrf/ros:melodic-desktop-full` , the first thing to do is to upgrade `cmake`  just follow the instructions [here](https://apt.kitware.com/) then patch the source code with this:

```diff
diff --git a/src/cpp/kiss_icp/3rdparty/tbb/tbb.cmake b/src/cpp/kiss_icp/3rdparty/tbb/tbb.cmake
index c9bc854..4c42802 100644
--- a/src/cpp/kiss_icp/3rdparty/tbb/tbb.cmake
+++ b/src/cpp/kiss_icp/3rdparty/tbb/tbb.cmake
@@ -34,8 +34,8 @@ ExternalProject_Add(
              ${ExternalProject_CMAKE_CXX_FLAGS}
              # custom flags
              -DTBB_BUILD_TBBMALLOC=ON
-             -DTBB_BUILD_SHARED=OFF
-             -DTBB_BUILD_STATIC=ON
+             -DTBB_BUILD_SHARED=ON
+             -DTBB_BUILD_STATIC=OFF
              -DTBB_BUILD_TESTS=OFF)
 
 # Simulate importing TBB::tbb for OpenVDBHelper target
```

I built it, test it, and it was actually working . I'm sorry but I will not probably provide support to make this work out of the box, if someone else needs support for ROS-MELODIC, just like this comment!

nachovizzo, 2023-02-23:
@martinlucan , @astronaunt This is now fixed in master automatically. Check the PR for details. If you have a modern cmake version installed, then you can build the project without having to modify the source.