PRBonn/kiss-icp #66: How to run dataloader kitti dataset

sonphambk, 2023-02-16:
Hi every one.
I want to run KISS-ICP with dataloader kitti. Could you show me the folder structure of kitti data?

TonyLiuWHU, 2023-02-16:
Here is my command and folder structure.
(kiss_icp) ➜  raw data kiss_icp_pipeline --visualize /home/username/Datasets/KITTI/odometry/Velodyne/00/velodyne

(kiss_icp) ➜  raw data tree /home/username/Datasets/KITTI/odometry/Velodyne/00/velodyne
/home/username/Datasets/KITTI/odometry/Velodyne/00/velodyne
├── 000000.bin
├── 000001.bin
├── 000002.bin
├── 000003.bin
├── 000004.bin
├── 000005.bin
├── 000006.bin
├── 000007.bin
├── ......

Then, we can see the window of 'RegistrationVisualizer' and the hint from kiss-icp.

nachovizzo, 2023-02-16:
The expected Latout is

```sh
Expected dataset layout
<data/kitti-odometry/dataset>
├── poses
│   ├── 00.txt
│   ├── 01.txt
│   ├── 02.txt
│   ├── 03.txt
│   ├── 04.txt
│   ├── 05.txt
│   ├── 06.txt
│   ├── 07.txt
│   ├── 08.txt
│   ├── 09.txt
│   ├── 10.txt
└── sequences
    ├── 00
    │   ├── calib.txt
    │   ├── poses.txt
    │   ├── times.txt
    │   └── velodyne
    ├── 01
    │   ├── ....
```

You can also check the eval folder where there is one full example on how to run it

closed 2023-02-16 by nachovizzo