PRBonn/kiss-icp #67: Rosbag convertion issue

berkaysahinaskar, 2023-02-16:
Thanks for this great job. 

I am trying your algorithm on kitti dataset for benchmarking and understand well how it works. I generally work on ros1 and kitti2bag tool is not converting point cloud good. 

After using that converted point cloud on kiss-icp it gives huge error when I evaluate.
[06_tl.pdf](https://github.com/PRBonn/kiss-icp/files/10754884/06_tl.pdf)

is there any way to correct that conversion?

nachovizzo, 2023-02-16:
Hello there, thanks for using our code.

I couldn't understand what is your experimental setup. The most important question would be, are you using ROS1 for evaluation? Why don't use the python API?

berkaysahinaskar, 2023-02-17:
No i just use ros1 for getting odometry and turning into poses. After that I use the evaluation code that KITTI provides.

nachovizzo, 2023-02-17:
I'm sorry, but I will not provide support for this. The reasoning behind this experiment is somehow wrong in my opinion:

```
Official data -> convert using non-official tools to another format -> run KISS-ICP -> Evaluate
```
On the other hand, we give you the chance (which was hardly possible before!) to just do:

```
Official data  -> run KISS-ICP (evaluation is automatic)
```

Additionally, I can't reproduce the setup since you haven't shared how to extract the poses, etc.

My humble advice: Use the right tools for the right tasks, ROS is not a scientific platform to evaluate algorithms, python is. Use python for experiments, ROS for deployment.

closed 2023-02-17 by nachovizzo

berkaysahinaskar, 2023-02-17:
Is the only difference in the result that one is written in ROS and the other is written in python? As a result, both the python and the ROS sides are working on Point Cloud. Is something different done when converting from KITTI to Point Cloud in Python? I'm trying to figure out if the problem is in the rosbag converter or the ROS.

nachovizzo, 2023-02-17:
Did you apply the calibration between the GT poses and the Velodyne frame?

berkaysahinaskar, 2023-02-17:
```
void kissICPCallback(nav_msgs::OdometryConstPtr odom){

    std::string myText;
    char myText2[10];
    std::cout.precision(17);
    std::cout << std::scientific;
    geometry_msgs::Pose pose;
    pose.position.z = odom->pose.pose.position.x;
    pose.position.x = odom->pose.pose.position.y;
    pose.position.y = odom->pose.pose.position.z;
    pose.orientation.w = odom->pose.pose.orientation.w;
    pose.orientation.x = odom->pose.pose.orientation.x;
    pose.orientation.y = odom->pose.pose.orientation.y;
    pose.orientation.z = odom->pose.pose.orientation.z;

    Eigen::Translation3d translation(pose.position.x, pose.position.y, pose.position.z);
    Eigen::Quaterniond rotation(pose.orientation.w, pose.orientation.x, pose.orientation.y, pose.orientation.z);
    // Eigen::Matrix4d matrix = (translation * rotation).matrix();
    ofs.precision(17);
    ofs << std::scientific; 

    ofs << rotation.matrix()(0,0) << " " << rotation.matrix()(0,1) << " " <<  rotation.matrix()(0, 2) << " " << translation.x() << " "
        << rotation.matrix()(1,0) << " " << rotation.matrix()(1,1) << " " <<  rotation.matrix()(1, 2) << " " << translation.y() << " "
        << rotation.matrix()(2,0) << " " << rotation.matrix()(2,1) << " " <<  rotation.matrix()(2, 2) << " " << translation.z() << " " << std::endl;
    
```
Here is the code that extract poses as KITTI pose format from odometry topic from ros. 

What kind of calibration do you use?

nachovizzo, 2023-02-17:
The gt poses are in the camera coordinate frame, not velodrome. I suggest you read the KITTI paper before continuing any further.