PRBonn/kiss-icp #68: Generating GT Poses from .PLY collection?

espackman-nv, 2023-02-16:
Hi all,
I saw a flurry of posts regarding custom datasets in KISS.

It got me thinking about a folder of .PLY files I have that can be visualized in KISS ICP, but no metrics are created because there are no GT poses available.

Is there a way to generate the poses without IMU data? Or some other way to get around this issue?

Thanks all.

nachovizzo, 2023-02-16:
Hello, what you are asking is not easy 😂.

If you want to evaluate the system you should have available another source for the poses that you know somehow is more accurate. In our community it's blindly accepted the fact that an expensive GNSS+IMU gives perfect poses and thus, can be used for evaluating the system.

If your question was on how to evuakte the kiss poses vs some ground truth you have, you can check the eval folder. The python API is quite easy and a 10 lines script should be able to give you the results.

Please correct me if I understood the question wrong.

Best!

closed 2023-02-16 by nachovizzo

espackman-nv, 2023-02-17:
Fair point, thank you!