PRBonn/kiss-icp #79: different results

Yuea922, 2023-02-24:
I run it with ros1, it seems that the latest version and v0.1.3 have different results
I play kitti bag 00 same as lio_sam
I don't know if it's my problem, but I didn't change any parameter
I didn't do an error evaluation, it's just that the trajectory looks different

v1.0.3
[image]

the latest version 
[image]

nachovizzo, 2023-02-24:
@Yuea922 I removed the kitti helper in the last release

The results are different because you are 1) not deskewing, 2) not compensating the calibration error in the dataset.

In principle, people should at least attempt to stop using KITTI on ROS, since I fail to understand why you guys are even doing this...

In any case, if you neglect the fact that running the experiments from python, or even the [python notebooks](https://github.com/PRBonn/kiss-icp/tree/main/eval) is easier, I've created a separate repository: https://github.com/nachovizzo/kiss_icp_kitti

closed 2023-02-24 by nachovizzo

Yuea922, 2023-02-25:
Thank you very much for your answer