PRBonn/kiss-icp #81: Developer Mode Install  Error Ubuntu 18.04

ntntnlstdnt, 2023-02-24:
Hello, 
I need assistance resolving the error below. I successfully installed the package, using `pip`, and ran a demo. Thank you 
```
lpw51@lpw51-ThinkPad-P51:~/kiss-icp/python$ pip install --verbose . 
Using pip 23.0.1 from /home/lpw51/.local/lib/python3.8/site-packages/pip (python 3.8)
Defaulting to user installation because normal site-packages is not writeable
Processing /home/lpw51/kiss-icp/python
  Running command pip subprocess to install build dependencies
  Collecting cmake
    Using cached cmake-3.25.2-py2.py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (23.7 MB)
  Collecting ninja
    Using cached ninja-1.11.1-py2.py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (145 kB)
  Collecting scikit-build
    Using cached scikit_build-0.16.7-py3-none-any.whl (79 kB)
  Collecting setuptools<66.0.0
    Using cached setuptools-65.7.0-py3-none-any.whl (1.2 MB)
  Collecting wheel>=0.32.0
    Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
  Collecting packaging
    Using cached packaging-23.0-py3-none-any.whl (42 kB)
  Collecting distro
    Using cached distro-1.8.0-py3-none-any.whl (20 kB)
  Installing collected packages: ninja, cmake, wheel, setuptools, packaging, distro, scikit-build
  Successfully installed cmake-3.25.2 distro-1.8.0 ninja-1.11.1 packaging-23.0 scikit-build-0.16.7 setuptools-65.7.0 wheel-0.38.4
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  running egg_info
  writing kiss_icp.egg-info/PKG-INFO
  writing dependency_links to kiss_icp.egg-info/dependency_links.txt
  writing entry points to kiss_icp.egg-info/entry_points.txt
  writing requirements to kiss_icp.egg-info/requires.txt
  writing top-level names to kiss_icp.egg-info/top_level.txt
  reading manifest file 'kiss_icp.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  writing manifest file 'kiss_icp.egg-info/SOURCES.txt'
  /tmp/pip-build-env-6sdzjsq0/overlay/lib/python3.8/site-packages/setuptools/config/expand.py:144: UserWarning: File '/home/lpw51/kiss-icp/python/' cannot be found
    warnings.warn(f"File {path!r} cannot be found")
  Getting requirements to build wheel ... done
  Running command Preparing metadata (pyproject.toml)
  running dist_info
  creating /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info
  writing /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/entry_points.txt
  writing requirements to /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/requires.txt
  writing top-level names to /tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/top_level.txt
  writing manifest file '/tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-modern-metadata-2a_jnvi6/kiss_icp.egg-info/SOURCES.txt'
  creating '/tmp/pip-modern-metadata-2a_jnvi6/kiss_icp-0.2.2.dist-info'
  /tmp/pip-build-env-6sdzjsq0/overlay/lib/python3.8/site-packages/setuptools/config/expand.py:144: UserWarning: File '/home/lpw51/kiss-icp/python/' cannot be found
    warnings.warn(f"File {path!r} cannot be found")
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: pydantic in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (1.10.5)
Requirement already satisfied: typer[all]>=0.6.0 in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (0.7.0)
Requirement already satisfied: tqdm in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (4.64.1)
Requirement already satisfied: pyquaternion in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (0.9.9)
Requirement already satisfied: natsort in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (8.2.0)
Requirement already satisfied: plyfile in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (0.7.4)
Requirement already satisfied: numpy in /usr/local/lib/python3.8/dist-packages (from kiss-icp==0.2.2) (1.24.2)
Requirement already satisfied: rich in /home/lpw51/.local/lib/python3.8/site-packages (from kiss-icp==0.2.2) (12.6.0)
Requirement already satisfied: click<9.0.0,>=7.1.1 in /home/lpw51/.local/lib/python3.8/site-packages (from typer[all]>=0.6.0->kiss-icp==0.2.2) (8.1.3)
Requirement already satisfied: colorama<0.5.0,>=0.4.3 in /home/lpw51/.local/lib/python3.8/site-packages (from typer[all]>=0.6.0->kiss-icp==0.2.2) (0.4.6)
Requirement already satisfied: shellingham<2.0.0,>=1.3.0 in /home/lpw51/.local/lib/python3.8/site-packages (from typer[all]>=0.6.0->kiss-icp==0.2.2) (1.5.0.post1)
Requirement already satisfied: typing-extensions<5.0,>=4.0.0 in /home/lpw51/.local/lib/python3.8/site-packages (from rich->kiss-icp==0.2.2) (4.5.0)
Requirement already satisfied: pygments<3.0.0,>=2.6.0 in /home/lpw51/.local/lib/python3.8/site-packages (from rich->kiss-icp==0.2.2) (2.14.0)
Requirement already satisfied: commonmark<0.10.0,>=0.9.0 in /home/lpw51/.local/lib/python3.8/site-packages (from rich->kiss-icp==0.2.2) (0.9.1)
[... 68 lines cut ...]
      raise SKBuildError(

  An error occurred while configuring with CMake.
    Command:
      /tmp/pip-build-env-6sdzjsq0/overlay/lib/python3.8/site-packages/cmake/data/bin/cmake /home/lpw51/kiss-icp/python -G Ninja -DCMAKE_INSTALL_PREFIX:PATH=/home/lpw51/kiss-icp/python/_skbuild/linux-x86_64-3.8/cmake-install/kiss_icp/pybind -DPYTHON_VERSION_STRING:STRING=3.8.16 -DSKBUILD:INTERNAL=TRUE -DCMAKE_MODULE_PATH:PATH=/tmp/pip-build-env-6sdzjsq0/overlay/lib/python3.8/site-packages/skbuild/resources/cmake -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3.8 -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python3.8 -DPYTHON_LIBRARY:PATH=libpython3.8.a -DPython_EXECUTABLE:PATH=/usr/bin/python3.8 -DPython_ROOT_DIR:PATH=/usr -DPython_INCLUDE_DIR:PATH=/usr/include/python3.8 -DPython_FIND_REGISTRY:STRING=NEVER -DPython3_EXECUTABLE:PATH=/usr/bin/python3.8 -DPython3_ROOT_DIR:PATH=/usr -DPython3_INCLUDE_DIR:PATH=/usr/include/python3.8 -DPython3_FIND_REGISTRY:STRING=NEVER -DCMAKE_MAKE_PROGRAM:FILEPATH=/tmp/pip-build-env-6sdzjsq0/overlay/lib/python3.8/site-packages/ninja/data/bin/ninja -DCMAKE_BUILD_TYPE:STRING=Release
    Source directory:
      /home/lpw51/kiss-icp/python
    Working directory:
      /home/lpw51/kiss-icp/python/_skbuild/linux-x86_64-3.8/cmake-build
  Please see CMake's output for more information.
  error: subprocess-exited-with-error
  
  × Building wheel for kiss-icp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /usr/bin/python3.8 /home/lpw51/.local/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_wheel /tmp/tmp4j_qwf6x
  cwd: /home/lpw51/kiss-icp/python
  Building wheel for kiss-icp (pyproject.toml) ... error
  ERROR: Failed building wheel for kiss-icp
Failed to build kiss-icp
ERROR: Could not build wheels for kiss-icp, which is required to install pyproject.toml-based projects

```

nachovizzo, 2023-02-24:
1. Ubuntu 18.04 end of life is in 2 months, so consider leaving it behind
2. `sudo apt install python3-dev` should solve the issue,

I haven't seen this issue because usually when you install `python3-pip` it should fetch python3-dev.

closed 2023-02-24 by nachovizzo

ntntnlstdnt, 2023-02-24:
Thank you for the prompt response.

YangSiri, 2023-02-27:
Hi, @nachovizzo  

Would cmake 3.10 work at KISS-ICP?

nachovizzo, 2023-02-27:
No, just install a new version is not difficult 👌