PRBonn/kiss-icp #89: ICP questions

Oreoluwa-Se, 2023-03-05:
Hello, this has been useful for the problem I've been working on. I have a few questions:

- Equation 13 for weighting outliers, the given equation in the paper is 0.5*e^2/(kt + e^2), but in the code, when calculating the Jacobian, the weight is calculated as e^2/(kt + e^2) any reason why?
- in the deskew process, there is a parameter "mid_pose_timestamp" I'm assuming this is doing something similar to the centroid estimation step when performing ICP with the SVD method.. is this the case?

nachovizzo, 2023-03-05:
Hello,

1. #35
2. This is a bit more complicated to explain, but basiaclly it's a parameter that represents in which coordinate frame you want to represent the deskewewd pointcloud. 0.0 means at the very beginning of the sweep, 1.0 would mean at the last measurement. We choose the middle

closed 2023-03-05 by nachovizzo

Oreoluwa-Se, 2023-03-05:
Got it. Makes sense! Thanks