PRBonn/kiss-icp #90: API to save the map

Alex-Beh, 2023-03-06:
Hello, 

thanks for the great project. May I know is there any way like ROS service to save the built map? I briefly searched through the codes but couldn't find it.

I can try to open an PR to add the feature.

tizianoGuadagnino, 2023-03-07:
Hi Alex,

our map is currently published as a simple ```PointCloud2``` msg in ROS. I think you can simply write a node that store the point cloud on disk using either PCL or Open3D. Consider that we use a local map, meaning that you will never obtain a global representation of the environment but just a clipped version of it around the current robot location.

I hope this helps and thanks for the interest in the project ;)

tizianoGuadagnino, 2023-03-09:
Because KISS-ICP is an odometry pipeline, not a SLAM system (no loop closures) so in principle we dont need the global map, as it will probably be inconsistent due to the drift.
Of course, feel free to extend the pipeline to allow for that if you want to give it a try. It should be enough to disable the ```RemovePointsFarFromLocation``` in VoxelHashMap. If you are successful come back to us with a PR. Thanks again for your interest.

horverno, 2025-02-17:
Alternatively, you can use this: https://github.com/jkk-research/pcd_publisher?tab=readme-ov-file#ros-2-pointcloud2-subscriber-writes-a-pcd-file