PRBonn/kiss-icp #91: Is it a bug in deskew function?

marrblue, 2023-03-07:
Hi, Thanks for your solid work and great contribution. 
I just found the parameter "mid_pose_timestamp" was set to be 0.5. I guess it should be 0.05 cause the interval time of each scan is 0.1s. 
[image]

jpaulos, 2023-03-07:
I'm not the author, but I believe what is happening here is that the timestamps themselves are already normalized to fall in the interval [0, 1] and reflect time progress from the beginning of the scan to the end. That is why a twist velocity doesn't appear at all and instead the delta_pose is directly being multiplied by "time". The "0.5" reflects the choice to deskew the scan to it's halfway time.

An advantage of this approach might be that it doesn't matter what the actual scan rate is (10Hz, 5Hz, 20Hz). Unfortunately for me I'd like to benchmark this deskewing process against deskewing using measured gyro rates, so the lack of "real units" is a bit confusing to work around.

closed 2023-03-07 by marrblue

nachovizzo, 2023-03-07:
@jpaulos thanks for the help! You answer is accurate indeeed

We've decided to simplify this module in this way with the obsession of reducing yet another parameter. But in older versions we compute the linear angle and angular velocity to compute the de skewing (you need to know the frequency of the lidar)

Even further, I've also experienced by exposing the velocities as inputs (I'm not sure if this was part of the open source release though, but I'm happy to send you the code)

If you check the kitti raw dataloader we already spit there the IMU velocities, for this same reason...to benchmark our approach against the IMU based. If you need guidance with this just let us know

@marrblue be aware that there is no assertion in the code, but we assume to receive a timestamp vector with normalized values, so if it's not between 0 and 1.0, stuff is going to get weird... Just FYI