PRBonn/kiss-icp #93: Using RealSense D435i

Genozen, 2023-03-10:
Hello, is it possible to use pointclouds generated from the d435i?

I see pointclouds in the visualized but doesn’t look like any points is being used so odometry estimate it quickly drifts away..

tizianoGuadagnino, 2023-03-10:
Hi @Genozen, we made some tests with the TUM [fr-desk](https://cvg.cit.tum.de/data/datasets/rgbd-dataset/download#) dataset and the pipeline was giving us reasonably good results even though the dataset contains RGB-D frames. So in principle should be possible to run KISS-ICP on your D435i, once you convert the RGB-D frame in a _PointCloud_.
Regarding the settings, you definetely need to tune the min and max range taking as reference the sensor datasheet (slightly larger for the min_range and slightly smaller for the max_range). This should be enough to get it to run.

Let me know if this helps, otherwise we can go more in depth with the settings. I'm curious to see which results will you obtain :smile:

Genozen, 2023-03-11:
@tizianoGuadagnino  Thanks for the swift reply. Currently, I have tried to decrease the max range a bit since the point cloud from the d435i is floating a bit the further it gets which I think it's contributing to some issues (camera pose drifts right away), but I will keep playing around with the ranges a bit.

Genozen, 2023-03-11:
D435i's stated range 0.28 m - 3.0 m

My setting to have decent ptclouds:
Max range: 6.5 m
Min range: 0.5 m

[image]

Still drifts a little and I think it really is because the fluctuations of the pointclouds near the further edges...odometry floats a bit but can snap back if it's at the position it has mapped before. For now I think this is good enough...

Orientation and translation seems off though

Genozen, 2023-03-11:
https://www.youtube.com/watch?v=EGLorSpi63Q

Genozen, 2023-03-11:
Here is what it looks like when mapping, in case anyone is interested...
https://www.youtube.com/watch?v=EGLorSpi63Q

Closing issue for now unless there's further suggestions on how to make it more robust (other than upgrading hardware)

closed 2023-03-11 by Genozen