PRBonn/kiss-icp #97: Mapping gives bad results sometimes

JereKnuutinen, 2023-03-18:
Hi,
I recorded two datasets using vlp 16 lidar. In the first dataset kiss-icp gives bad mapping results as shown below:
[image]

However when I used kiss-icp in the second dataset it perfomed well:

[image]

Why is this happening. Since both datasets are nearly identical as can be seen from the images.

fratopa, 2023-03-18:
From simulations I noticed that kiss-icp tends to perform quite poorly when dealing with high speed movement and when computationally constrained. If you can't find the problem and some runs go very well and other very poorly check htop to see if resource utilisation may be the problem. If that's the problem reducing the max range and thusly the local map size will help you reduce the load

JereKnuutinen, 2023-03-18:
Thank you for your comment. What do you mean by ”computationally constrained”?

fratopa, 2023-03-18:
In my case it manifests as 100% utilisation on all cores, that's the one thing that was similar between all my poorly performing runs

JereKnuutinen, 2023-03-19:
I monitored core utilization and only 2-3 cores out of 12 were at 100% utilization. So that is not reason for the bad mapping result.

fratopa, 2023-03-19:
That's very strange, I find the load to be evenly spread, regardless of utilisation. Maybe we can have some inputs from the creator. What are your settings?