PRBonn/kiss-icp #106: How to add CUDA support?

tiesus, 2023-03-28:
Hi,
first of all I really appreciate your work. Your method is working with minimum configuration on my data and produces awesome results in most cases.

One downside is the quite high cpu load. Therefore I would like to ask if the are any plans to support CUDA to shift cpu load to the gpu?
If you don't plan to add CUDA are there any hints how start with it?

ahans, 2023-03-28:
I believe there's also quite some potential to speed up the CPU version. For example, when downsampling the point cloud, it is copied at least once, maybe even twice. Doing those operations in place would probably help. Another option could be not modifying point clouds at all, but working with index lists instead. For the parallel operations, on top of parallelization via threads as it is done currently, SIMD vectorization would probably help.

Since there are a lot of parallel operations, a port to CUDA would probably be straightforward as well. I'm not sure about the memory access patterns for the voxel hash map. One must be careful to avoid a lot of cache thrashing, GPUs are even more sensitive about this than CPUs. So maybe some other data organization would be required there. But as a first experiment a straight port could do.

However, I wouldn't start any of those optimizations without 1) having some unit tests in place and 2) setup some benchmark application that allows for meaningful A/B comparisons. Looking at `top` to get learn if a change helped or not won't get you very far.

tiesus, 2023-03-28:
Thank's for your response. I will have a look at the preprocessing as starting point and write some comparsion methods to evaluate the success.

nachovizzo, 2023-03-29:
Hello, @tiesus Thanks for the interest in the project.

Indeed, what @ahans is pointing is 100% correct. Without any test/benchmark, starting this effort is kind of a dead end, since there are no instrumentation in the code to analyze how fast/slow it is. Since this is a research project, it will be very unlikely that we add this in the future. What we do have right now is the `eval` folder, containing some scripts to run the evaluation in multiple datasets. But the amount of data you need to download and the time it takes, is.. Too much, let's say!

If you would like to CUDAize some parts of the code, you could start by inspecting the single thread branch [here](https://github.com/PRBonn/kiss-icp/tree/nacho/single_thread_implementation), you can also `git diff` and check the portions we've managed to parallelize so far using TBB.

The bottleneck of the pipeline is the nearest neighbor search (as most other pipelines) So I'd rather start from there than trying to parallelize the preprocessing module (is not efficient, but it's not the slowest part of the system). To do so, you first need to create a nice hash-table in CUDA (or use any existing one), the overall effort is not "that" easy.

 @gisbi-kim also did some analysis on [Youtube](https://www.youtube.com/watch?v=Yl3CytRfXXU&)

Be aware that by no means, the parallel code is "proper" code, I guess if some hpc folks see this, they would cry. So don't use this as a "good" example/

closed 2023-04-14 by tizianoGuadagnino