PRBonn/kiss-icp #108: Adding Time field to pointcloud2 to use Deskew and  odometry results Plotting

MblRobotics, 2023-03-28:
Hello 
First , thank you For such great Package
I'm running this package on a bag file recorded from industrial AGV and i want to run kiss-icp on it with deskew enabled.
The pointcloud2 message output of the bag file doesn't contain time or t or timestamp field so i added it manually but i don't know how to fill the field and with which data 
Currently , the package runs (despite i didn't fill any data to the time field ) but i don't know if the deskew is giving a result or not(i feel it's the same result whaterver deskew is true or false) also i changed the utils.hpp file to print time field data and i see some informations so isupposed that it's filled automatically but i'm not sure of it
Also i want to plot the odometry output , i tried to use rqt_multiplot or rqt_plot but i'm unable to choose the odometry topic, so could you please help me with this 
Thank you so much

nachovizzo, 2023-03-30:
Hello, thanks for using our package

For us is always hard to help without accessing the data. What you should know is that the deskewing method expects timestamps between 0.0 and 1.0. If we detect that the timestamps are on epoch time then we will normalize it.

The easiest way to test this (if you are testing offline) is to dump the clouds to the disk and create a custom dataloader with python that adds the timestamps as you wish

MblRobotics, 2023-03-30:
Thank you sir for your response , can you give me an example of dataloader

closed 2023-03-30 by nachovizzo