PRBonn/kiss-icp #111: Investigate ROS 1 build after ROS 2 integration

nachovizzo, 2023-03-31:
After merging #74, while building the ROS 1 node, instead of the typically `install` msgs I see this now:
```sh
[kiss_icp:make] [100%] Linking CXX executable /home/user/ros_ws/devel/.private/kiss_icp/lib/kiss_icp/odometry_node
[kiss_icp:make] [100%] Built target odometry_node
[kiss_icp:symlink] Symlinking /home/user/ros_ws/devel/share/kiss_icp/cmake/kiss_icpConfig.cmake
[kiss_icp:symlink] Symlinking /home/user/ros_ws/devel/share/kiss_icp/cmake/kiss_icpConfig-version.cmake
[kiss_icp:symlink] Symlinking /home/user/ros_ws/devel/lib/pkgconfig/kiss_icp.pc
[kiss_icp:symlink] Symlinking /home/user/ros_ws/devel/lib/kiss_icp/odometry_node
```

The ROS1 node seems to be working, but maybe there is something I'm missing, or something that might fire back in the future.

nlamprian, 2023-08-09:
As long as you use the install space (`catkin config --install`), the file will be copied there in the end.

```
-- Installing: /root/catkin_ws/install/lib/kiss_icp/odometry_node
```

The CMakeLists.txt looks fine.

```
root@30d53bb2e4cd:~/catkin_ws# tree install/share/kiss_icp/
install/share/kiss_icp/
|-- cmake
|   |-- kiss_icpConfig-version.cmake
|   `-- kiss_icpConfig.cmake
|-- launch
|   |-- odometry.launch
|   `-- odometry.launch.py
|-- package.xml
`-- rviz
    |-- kiss_icp_ros1.rviz
    `-- kiss_icp_ros2.rviz

3 directories, 7 files
root@30d53bb2e4cd:~/catkin_ws# tree install/lib/
install/lib/
|-- kiss_icp
|   `-- odometry_node
`-- pkgconfig
    |-- catkin_tools_prebuild.pc
    `-- kiss_icp.pc
```

nachovizzo, 2023-08-15:
@nlamprian Thanks a lot for shedding some light. As usual, it was just being paranoid.

Cheers!

closed 2023-08-15 by nachovizzo