PRBonn/kiss-icp #116: Migrate to nanobind

nachovizzo, 2023-04-03:
[nanobind](https://github.com/wjakob/nanobind/discussions/89) looks quite promising. Specially due to the support of ndarrays, etc.

I've tried a draft, but failed to add support for Eigen. There is an [ongoing discussion](https://github.com/wjakob/nanobind/discussions/89) about this problem.

The idea of this issue is to 
- [ ] Analyze if nanobind is a good replacement for pybind11
- [ ] Analyze if it makes life simpler for adding CUDA support(#106)
- [ ] Get rid of all the Eigen hacks (look inside pybind folder)
- [ ] Hopefully simplify the python bindings of the project

jpaulos, 2023-04-03:
My impression is that the origin of some of the pybind11 awkwardness relates to working with std vectors of Eigen vectors, as opposed to just an Eigen array. It seems like the new Open3D Tensor API is moving away from that strategy in an effort to make conversions between Numpy arrays, Eigen arrays, and their PointCloud Tensor data more transparent and more often shared memory. Does that seem accurate?

nachovizzo, 2023-11-02:
Move issue to backlog due inactivity

closed 2023-11-02 by nachovizzo