PRBonn/kiss-icp #122: poses vector grows indefinitely

ahans, 2023-04-04:
The [poses vector](https://github.com/PRBonn/kiss-icp/blob/main/cpp/kiss_icp/pipeline/KissICP.hpp#L78) is only ever added to. While poses are small objects, eventually any system will run out of memory. We need some bound for this.

To fix this, we could define some number of last poses that are kept. However, it looks like currently only the very first and last two poses are ever used. An even better fix might therefore be just keeping those.

tizianoGuadagnino, 2023-04-14:
Hi @ahans,
The unbounded growth of the pose vector it is not a sensible problem in our point of view. A pose using Sophus is represented by 7 doubles. Unless there are specific memory restrictions, you need an extremely high number of poses for this to be a real problem. Furthermore, we need to store the all trajectory so that it can be inspected after a run.

ahans, 2023-04-14:
Keeping all poses is just nothing we need in KISS-ICP core. If you want to inspect them after a run, there should be another place where you accumulate the poses. Maybe somewhere in the Python driver. I was under the impression you wanted to do things properly in this project, and unbounded growth is simply tech debt. And in the present case it's actually quite simple to fix it, as shown in my PR #124. Granted, the Python side of things would need some work, but it's still small enough to not accept this. But all of that is IMHO. If you want to keep this as is, fine.

commit d5132999 (2024-04-12): Refactor KissICP pipeline to mantain just 1 pose instead of full trajectory. Fixes #122 (#318)

* First draft for fixed_size_vector. Fixes #122

* Pose pair, many ifs go Rhaus if this works

* New adaptive threshold based on pose vector modification

* Now has moved also rhaus from python

* Push new draft with no new types

* rename

* Fix ROS2

* Remove unused functions from cpp

* Reduce code lines more

* Change names and add some more comments

* Marry python and C++ implementations

* remove unused headers

* Keep copy-pasting 🤦

* Move poses array one level up, and add new deskew functi [... cut]