PRBonn/kiss-icp #125: How to bias the estimation towards "flat world"?

ahans, 2023-04-04:
When using KISS-ICP to estimate motion of a ground vehicle, how can I compensate drift in z and pitch and roll using the assumption that the vehicle travels on flat ground? For now, I have hacked in some code that lets the z-component of the translation as well as the rotation components about x and y axis tend towards zero. I apply this to each pose after the `RegisterFrame()` step. Since this is probably quite a hack, I wonder if there's a more principled way of doing something like this?

nachovizzo, 2023-04-05:
I guess this is to restrict the 6D state estimation problem to a 3D (x, y, yaw). If the robot has no degree of freedom on the Z axis then the pitch and the roll angles should not be considered.

I guess the "propper" way of doing this is rewriting the optimization to optimize for a 3D pose instead of 6D. Not sure if there is a nicer "higher level" way of doing this.

ahans, 2023-04-05:
Well, my "robot" is a car, so technically it has 6 DoF. It's just that I want to somewhat penalize z, pitch, and roll being not quite small. Could we add some sort of regularization to the optimization that would favor small z, pitch, and roll?

Another idea I have is using the output of KISS-ICP as measurements for something like a Kalman filter where I can then hopefully formulate my prior knowledge as part of the motion model or encode it via covariances.

tizianoGuadagnino, 2023-04-14:
Hi @ahans,

The idea of the regularization in the optimization is definitely the simpler solution, and probably is gonna do what you want. I will actually just put the regularization on the z and roll, as the car can definitely have a pitch when is moving uphill. If you need help to what to modify, let me know i'm happy to help. For now I close the issue.

closed 2023-04-14 by tizianoGuadagnino