PRBonn/kiss-icp #128: Kiss-ICP struggles with rotation around X or Y axis and translations on Z axis ?

YohanFT, 2023-04-06:
Hi,

First of all, thanks for this great work !

I noticed that when I perform rotations around the X or Y axis or translations following the Z axis, KISS-ICP struggles and often do nothing or detect wrong transformations instead. 

I'm currently working indoor with an Ouster OS0 with 32 beams and a 6 meters max range threshold. I didn't find anything that could help me with this since most works/issues on LiDAR are about a fixed LiDAR that is rarely rotated around the X or Y axis.

On this example, the bigger pose is KISS-ICP output while the smaller one comes from a visual slam which can be considered as ground truth here.
 
[image]

Do you have any clue on what could be happening ?

jpaulos, 2023-04-06:
I also had trouble with the OS0-32 configuration specifically (other sensors were fine). I chalked it up to it's extremely sparse beam pattern and didn't investigate more deeply. The sensor has a 90 degree vertical field of view and only 32 scan lines. I became suspicious that this was a problem especially at the beginning, before the map has been densified by motion.

nachovizzo, 2023-04-12:
So, this is a very challenging dataset!. Thanks for sharing the data with us.

## Some analyses on this data

As a matter of fact, you will need to tweak the system to tune it for your own needs. This is something we can't on your name since it's not our business :+1:. The best approach would be to read the paper, understand the components of the system and read the implementation. We spend extra time trying to make the implementation accessible for understanding, so we hope not to induce pain during this process (as the usual workflow)

The second step would be to understand the type of data you are dealing with. It's a quite "sparse" 32-beam sensor. So KISS-ICP typically does not perform well out-of-the-box, since we assume we can build on-the-fly a semi-dense (let's call it) map representation we can align new frames. To also speed up the runtime we *heavily* downsample the clud, which  in your particular case **does not make sense!!**. Since the cloud is already extremely sparse (+ the indoor scenario). I'd suggest using the python package to debug the system, and then try to replicate this on the C++/ROS side. It's much more convenient. For example, if you visualize the pipeline and press "K" you can see the (almost unexisting) keypoints.

In summary, you have a challenging scenario (not the easy ground-robot that does not move a lot) and for this you would need to spend extra effort to make a system like ours work.

## One possible solution

**Nevertheless!** (besides me being grumpy ;) ) I've also come with a “solution” for -at-least-the-data-i-had. 

1.  Disable dowsmapling for the system! It's too sparse! This you can easily achieve it in the python side like this (doing this on the C++ is exactly the same):
```diff
diff --git a/python/kiss_icp/kiss_icp.py b/python/kiss_icp/kiss_icp.py
index 6bcdae0..6dc1afb 100644
--- a/python/kiss_icp/kiss_icp.py
+++ b/python/kiss_icp/kiss_icp.py
@@ -74,8 +74,7 @@ class KissICP:

     def voxelize(self, iframe):
         frame_downsample = voxel_down_sample(iframe, self.config.mapping.voxel_size * 0.5)
-        source = voxel_down_sample(frame_downsample, self.config.mapping.voxel_size * 1.5)
-        return source, frame_downsample
+        return frame_downsample, frame_downsample
```
2. Change some default parameters. In my case, I've created a config.yaml file like this one:

```yaml
data:
  min_range: 0.0

mapping:
  voxel_size: 0.5
```

3. Run! `kiss_icp_pipeline whole_seq_only_point_cloud.bag --config config.yaml -topic /point_cloud/lidar --visualize`

You can try this by cloning this repo and just hitting make (or also `pip install -U kiss-icp` and then modify the code in place)

https://user-images.githubusercontent.com/21349875/231468122-1ab80f3a-06a1-4014-a49c-a4268e7d055c.mp4

[image]
[image]

closed 2023-04-12 by nachovizzo

YohanFT, 2023-04-14:
Thank you very much for both answers. I've been able to reproduce easily ! I'll investigate with another LiDAR as well

harisbinyousaf24, 2023-11-06:
Hi @nachovizzo and the team Thanks for creating this masterpiece and making life easier!!
Although the pipeline ran perfectly well in my case but I need to know how the ground truth poses need to be input..like I am giving path to my bag file and visualize it but after the visualization has run I get Warning no GT Poses available. Does GT poses need to be present in the bag file ? I have GPS data containing Latitude, Longitude and Altitude can i use them as my ground truth? if so how?

SangHunHan92, 2023-11-15:
I also have GPS data. Can I use kiss-icp using this? How is this possible?

I am in an environment where the pitch changes periodically (I scan the building by shaking the Lidar up and down). Therefore, it is necessary to supplement it using GPS.