PRBonn/kiss-icp #135: Merge mcap and ROS [1/2] dataloaders

nachovizzo, 2023-04-07:
I guess `rosbags` will soon improve the support for reading `mcap` files, and therefore, after we merged the ros dataloaders into one unique (#134) it make sense to merge these two dataloaders which are almost identical

Rajat-Arora, 2023-08-25:
Hi @nachovizzo,

nachovizzo, 2023-08-25:
@Rajat-Arora hello. Thanks for the interest.

The first steps I'd take is amazing the python Dataloaders and understanding what are the main differences and reason how to combine those in a smart way. I'd get some ROS 1 and 2 datasets and a mcap one to test the logic.

Rajat-Arora, 2023-08-25:
Thanks for the headers,
1. I will get some ROS1, 2 and mcap datasets and test/run the Python version of the algorithm to run on it.
2. See how the Python data loader responds to each one of them and then think of a way to combine it.

PS: Please point if I am wrong in somewhere and will keep u posted :)