PRBonn/kiss-icp #138: Weird results using livox lidars

rsc9421, 2023-04-10:
HI, im trying kissicp with livox lidar mid70 in a gazebo simulation

Here im not able to put images and files so i upload it to my drive, in that folder you can find:
* ros2 rosbag with livox lidar
* in order to compare the generated pose and the gazebo pose i put a .npy files with gazebo poses
* the command that i use is: 
` kiss_icp_pipeline --topic /mid70 --visualize rosbag2_2023_04_10-11_24_36 --config kissicp_config.yaml `
* in the folder you can find the config file

https://drive.google.com/drive/folders/13uQvY6y-bCEAL9__BWWQ9h2hPRkHCOLg?usp=share_link

rsc9421, 2023-04-11:
Hi nacho, yes it is a real need, in the company where i work we have a robot with 6 lidars, that's why we are very interested in kiss-icp :) 

i also tested in a real world rosbag using 1 lidar mid70, with the robot moving in forward direction, and i compare it with livox_mapping, here are the results:
* test with kiss-icp (default config)
[https://www.youtube.com/watch?v=9THVQBYRaNs](url)
* test with livox_mapping (default config)
[https://www.youtube.com/watch?v=9THVQBYRaNs](url)
* rosbag used:
https://drive.google.com/drive/folders/1bRMD20lA0FXbFH-K9qg9WL7kQlA93Lcl?usp=sharing

as you notice there is a weird result with kiss icp, but if i change the voxel size to 0.1m i get much better results, but at the cost of consuming 6 times of cpu usage, there's a plot in the drive comparing results:
* kiss-icp with voxel_size=0.1m => mean=647% (6 cores of cpu at 100% + 1 core at 47%)
* kiss-icp with voxel_size=0.5m => mean=97% (1 core of cpu at 97%)
* livox_mapping with voxel_size=0.1m => mean=80% (1 core of cpu at 80%)

since we have 6 lidars in the robot it is almost impossible to apply kiss-icp in each lidar, could you suggest us some option to speed up the processing?. Perhaps adding support for GPU CUDA would make a big difference compared to other visual odometry algorithms

Finally, I have other real-world rosbags with the livox lidars, if you want I can share them

nachovizzo, 2023-04-11:
Hey Casimiro. Thanks a lot for sharing the data. (This also relates #51)

Livox lidars are always challenging to use without another external source. Anyway, for this particular small bagfiles, I think this set of parameters works a bit better:

**config.yaml**:
```yaml
data:
  deskew: False
  preprocess: False

mapping:
  voxel_size: 0.1
  max_points_per_voxel: 10

adaptive_threshold:
  initial_threshold: 2.0
  min_motion_th: 0.1
```

Results:
[image]
[image]

On my **DESKTOP** cpu this is running at 50FPS. I think the Livox is spitting data at 30FPS, so yes. This is actually slow for the application. There is an ongoing discussion on using CUDA (#106) but I don't think this will fix the problem in your case. Since you can speed it up, but using more cores, not doing something different. This basically means more energy, etc. Plus, the accelerators are typically also used for other applications (object detection, etc) If you want you could check the `nacho/single_thread_implementation` branch where there is NO multithreading (in case your energy requirements ask for it)

The other solution would be to take the initial guess from another sensor and try to reduce the number of iterations of the ICP.

tiesus, 2023-04-17:
I investigated further to speed up kiss icp with livox lidar. There is a huge speed up if you deskew the pointcloud with the livox undistortion package, before feeding them into kiss icp (https://github.com/Livox-SDK/livox_cloud_undistortion). I didn't measure the speedup, but i think it's factor 2. We can run kiss icp now without issues.