PRBonn/kiss-icp #144: request for a pyhton demo: register two random point clouds

yanconglin, 2023-04-14:
Hi kiss-icp researchers,
Thank you for the excellent work!
Can you please share a python demo on registering two custom frames? Simply speaking, the input would be two point clouds (source/target) and the output is the transformation that maps source to target.

Many thanks!

tizianoGuadagnino, 2023-04-14:
Hi @yanconglin ,

What you are asking is essentially a point2point ICP implementation which is not restricted to the LiDAR odometry domain. KISS-ICP is specifically meant for odometry estimation, so it doesn't make sense to use two "random" point clouds. You can have a look [here](http://www.open3d.org/docs/release/tutorial/pipelines/icp_registration.html#Point-to-point-ICP).

closed 2023-04-14 by tizianoGuadagnino

yanconglin, 2023-04-18:
hey, thank you for the reply. 
A clarification: I was trying to register two random frames from the same sequence.
But I think the problem is somehow solved?  The code is also attached (adapted from kiss_icp). Might be useful to other people.

`
class kiss_icp_register:
    def __init__(self, config_file):

        config = load_config(config_file, deskew=False, max_range=None)
        self.poses = []
        self.config = config
        self.adaptive_threshold = get_threshold_estimator(self.config)
        self.local_map = get_voxel_hash_map(self.config)
        self.preprocess = get_preprocessor(self.config)

    def register_frame(self, frame1, frame2):
        self.local_map.empty()
        self.local_map.clear()
        self.local_map.update(frame2)

        # Preprocess the input cloud
        frame1_tmp = self.preprocess(frame1)

        # Voxelize
        source, frame_downsample = self.voxelize(frame1_tmp)

        # Get motion prediction and adaptive_threshold
        sigma = self.get_adaptive_threshold()

        # Compute initial_guess for ICP
        prediction = self.get_prediction_model()
        last_pose = self.poses[-1] if self.poses else np.eye(4)
        initial_guess = last_pose @ prediction

        # Run ICP
        new_pose = register_frame(
            points=source,
            voxel_map=self.local_map,
            initial_guess=initial_guess,
            max_correspondance_distance=3 * sigma,
            kernel=sigma / 3,
        )
        frame1_icp = transform_points(frame1, new_pose)
        # self.adaptive_threshold.update_model_deviation(np.linalg.inv(initial_guess) @ new_pose)
        # self.local_map.update(frame_downsample, new_pose)
        # self.poses.append(new_pose)
        return frame1_icp, frame2, new_pose

    def voxelize(self, iframe):
        frame_downsample = voxel_down_sample(iframe, self.config.mapping.voxel_size * 0.5)
        source = voxel_down_sample(frame_downsample, self.config.mapping.voxel_size * 1.5)
        return source, frame_downsample

    def get_adaptive_threshold(self):
        return (
            self.config.adaptive_threshold.initial_threshold
            if not self.has_moved()
            else self.adaptive_threshold.get_threshold()
        )

    def get_prediction_model(self):
        if len(self.poses) < 2:
            return np.eye(4)
        return np.linalg.inv(self.poses[-2]) @ self.poses[-1]

    def has_moved(self):
        if len(self.poses) < 1:
            return False
        compute_motion = lambda T1, T2: np.linalg.norm((np.linalg.inv(T1) @ T2)[:3, -1])
        motion = compute_motion(self.poses[0], self.poses[-1])
        return motion > 5 * self.config.adaptive_threshold.min_motion_th

    def visualize(self, seq_0, seq_1):
        seq_0 = np.concatenate([seq_0, np.zeros((len(seq_0), 1))+1], axis=1)
        seq_1 = np.concatenate([seq_1, np.zeros((len(seq_1), 1))+2], axis=1)
        visualize_pcd(np.concatenate((seq_0, seq_1), axis=0))

`