PRBonn/kiss-icp #146: Getting bad results when lidar at small angle

JereKnuutinen, 2023-04-19:
Hi,

I get some bad results when my lidar (vlp-16) is at small angle, about 15 degrees. As you can see from figure: 
[image]

When my lidar is not at angle, the results are good.

I also attach my ros2bag if the authors want to take a look. The bag contains multiple topics but two needed topics are synced_pc_lidar_1 and synced_pc_lidar_2. The data from synced_pc_lidar_1 is the data that performs well and the data from synced_pc_lidar_2 topic does not perform well. 

https://drive.google.com/drive/folders/1VwJ2gjFhztBUoMdh61DKC_FGEUTrwLz3?usp=sharing

In addition, I am using the following parameters:

data:
  deskew: False
  preprocess: True
  max_range: 20.0 # can be also changed in the CLI
  min_range: 1.5

mapping:
  voxel_size: 0.1 # <- optional
  max_points_per_voxel: 20

adaptive_threshold:
  initial_threshold: 2.0
  min_motion_th: 0.1

benemer, 2023-04-20:
Hi,

Please check out the closed issues, we already investigated the challenges of sparse sensors. See [this reply](https://github.com/PRBonn/kiss-icp/issues/128#issuecomment-1505258331) for example.

I just removed the second downsampling as described by Nacho and used the following config:

```yaml
data:
  max_range: 20.0
  min_range: 1.5

mapping:
  voxel_size: 0.5
```

and obtained these results:

**synced_pc_lidar_1**
[image]

**synced_pc_lidar_2**
[image]

JereKnuutinen, 2023-04-21:
Thank you! Another question. In the code side what can I do if I dont care performance or execution time, I am only interested in how accurate pose eatimate is. In other word, have you made some compromises regarding on accuracy so that real time performance is met.

JereKnuutinen, 2023-04-22:
Besides downsampling is there anything that increase runtime but would also increase accuracy of pose estimates?

benemer, 2023-04-24:
This is hard to say in general. Our goal was to design a pipeline that realizes the most important components to work on a large variety of datasets. I guess there is definitely room to increase the accuracy on certain data, but this usually requires extending the approach or tuning the method on a specific target domain.

closed 2023-04-24 by benemer