PRBonn/kiss-icp #148: Struggling with robustness in rotations

jediofgever, 2023-04-25:
Hi, 

I have tried KISS-ICP on a simulated Ouster-64, and it seems that in rotations the system struggles. 
I have seen previous issues such as; https://github.com/PRBonn/kiss-icp/issues/128 and https://github.com/PRBonn/kiss-icp/issues/17 

These mainly point to sparsity issues but In my I case played with `voxel_size`  in the range of (0.2 - 1.0) and I doubt Ouster-64 would be considered a sparse sensor. 
Though, With linear motion, the system works as expected. 

I will attach a simple ros2 bag file and video for your reference, I would appreciate it if you could take a look and reproduce it. 

Note that I tried with online data as well as data recorded in a bag file, with the same result. 

https://user-images.githubusercontent.com/18460336/234314616-db0ebd19-eb76-4337-a12c-399bfaf466af.mp4

jediofgever, 2023-04-26:
Many Thanks @nachovizzo ! , 
I will investigate the simulator further, though the FPS of LiDAR data is reasonable, there might be some other issues with physics parameters of the robot model.

nachovizzo, 2023-04-26:
Ok, good luck! Feel free to comment on your investigations