PRBonn/kiss-icp #150: correct usage format for generic datasets

naveenmoto, 2023-05-01:
Trying to use
kiss_icp_pipeline <folder_path_with_xyz_pointcloud>
gives an error

Trying to guess how to read your data: "pip install kiss-icp[all]" is required
[ERROR], File format not supported

Each xyz file is named frame_id.xyz and has the pointcloud as follows:
2.171924352645874 -29.2801513671875 9.367907524108887
2.5721511840820312 -34.6786003112793 10.233442306518555
2.7723987102508545 -37.379634857177734 9.176665306091309
....
...

What would be the right format to use kiss-icp?

benemer, 2023-05-02:
Hey,

can you send us a sample file? How do you usually load these types of files?

naveenmoto, 2023-05-02:
Hey Benemer - thanks for checking back

I converted the raw point clouds from our Ouster dataset into xyz format and save them as <frame_id.xyz>. Attaching a sample. I put them all in a folder and pointed kiss icp to the folder

kiss_icp_pipeline <folder_path_with_xyz_pointcloud>

[20.xyz.zip](https://github.com/PRBonn/kiss-icp/files/11372586/20.xyz.zip)

benemer, 2023-05-02:
Can you please check your version of KISS-ICP with `kiss_icp_pipeline --version`? You need version >= 0.2.9

naveenmoto, 2023-05-02:
It says 

KISS-ICP Version: 0.2.9

Should this version work OK?

benemer, 2023-05-03:
Yes, that works. In that case you probably don't have the extra dependencies like trimesh installed. Please install KISS-ICP with `pip install "kiss-icp[all]"`.

closed 2023-05-05 by tizianoGuadagnino