PRBonn/kiss-icp #151: Failure case study

yanconglin, 2023-05-01:
Thank you for releasing the tool.
Recently I am tryin KISS_ICP on waymo for ego motion estimation. However, I find several failure case where the model goes substantially wrong. Attached please find an example. Could you please tell me how to fix this issue (e.g., picking the correct parameters)?

The figure shows two registered frames (colored in blue and green). As you can see the the non-ground points do not align. 
[image]

benemer, 2023-05-02:
Hey,

this is hard to say without having the data at hand. Can you specify which data you used?

Are these the first two frames of the sequence? Is the car already in motion?

yanconglin, 2023-05-02:
yes, I have shared the source with you. 
Here is a mini example: [kiss_icp_waymo_failure](https://drive.google.com/drive/folders/1C_b1mIvIVKIXdxVcJiyCNLUEMyeSf0DT?usp=sharing)

benemer, 2023-05-02:
Thank you!

I think I found the problem. When estimating the first relative pose, we do not have a good initial guess since our prediction model requires at least two estimated poses (assuming no other sensors like IMU or wheel odometry). Therefore, we initialize our estimation with an identity pose. KISS-ICP finds a local minimum by aligning the scan lines as you can see in your image. We found that KISS-ICP usually recovers from this after some frames (check for example KITTI odometry sequence 12). In your case, 5 frames is probably not enough. Have you tried running it on the whole sequence?

A fast way to improve this is to increase the `initial_threshold` parameter. If you for example set it to 10, you will 1. keep all correspondences and 2. reduce their suppression by the robust kernel. Using the default parameters and increasing the initial threshold to 10, the pipeline already recovers after a couple of frames:

https://user-images.githubusercontent.com/38326482/235620071-52cb452f-ebf3-4910-a861-44c10f72ca9f.mp4

However, these are rather edge cases for which you might need to find a more customized initialization phase or the integration of other sensors.

yanconglin, 2023-05-02:
many thanks for the reply.  Strange, increasing "initial_threshold: 10.0" does not help on my local machine. 
Can you please share with me the config file.

benemer, 2023-05-02:
Are you using our pipeline and visualizer? I did not use the code you sent me except the data loading part to convert the point clouds to `ply` files. I used [this](https://github.com/PRBonn/kiss-icp/blob/main/config/basic.yaml) config with only the `initial_threshold` parameter changed.

yanconglin, 2023-05-02:
I am using class [KISS_ICP](https://github.com/PRBonn/kiss-icp/blob/1b0529169b2461272a97e959e2a9deea57dbb661/python/kiss_icp/kiss_icp.py#L34), with my own datalodaer which directly takes as input the given '.npz' file as indicated in the code snippet.

benemer, 2023-05-02:
In the snippet you sent me you are importing `KissICP` but using the `egomotion` class instead.

yanconglin, 2023-05-02:
well, I renamed "KISS_ICP" class to "egomotion" class  for the clarity (for the project I am working on).  The "egomotion" class is simply a copy of KISS_ICP" class. I do not want to alternate the KISS_ICP codebase. sorry for the confusion

benemer, 2023-05-02:
Ah okay! I did not compare if it's exactly the same.

benemer, 2023-05-03:
You are passing the wrong timestamps. This is supposed to be a `N x 1` numpy array of the individual timestamps for each of the `N` points in the frame.

closed 2023-05-09 by benemer

yanconglin, 2023-05-09:
Hi, again,  I would like to ask if KISS-ICP is deterministic, as I obtained slightly different results on the same sequence. I wonder if there is a way to fix the randomness or am I making a mistake? This would be great for debugging, since KISS-ICP is the very first step of a complicated framework.

benemer, 2023-05-16:
What kind of randomness are you observing, so how much does it affect the results?

yanconglin, 2023-05-17:
Hi, I am using KISS-ICP for ego-motion estimation. each time it outputs a slightly different result. Currently, I am debugging my model, which requires a deterministic pose from ego-motion estimation.

benemer, 2023-05-21:
I moved this to a new issue since this is already closed, see https://github.com/PRBonn/kiss-icp/issues/165