PRBonn/kiss-icp #152: Inconsistent and worse results in ROS launch

Shahrullo, 2023-05-02:
Thank for sharing `kiss-icp`. It is very easy to use and works really well on PC. However, there are two problems when I run the model in Jetson Orin AGX: 

1. In ROS, it is giving very bad results. Additionally, it shows different results in each run. 

**First run**
[image]
**Second run**
[image]

2. In Python run, the results are very good and consistent in each run. However, it is very slow compared to the ROS. Usually 2-3 times slower. How can I improve the speed in Jetson Orin AGX?   

[image]

In both cases, the same rosbag file is used.

tizianoGuadagnino, 2023-05-03:
Hi @Shahrullo, I think the ROS side might have a frame-dropping problem. That will explain why you get consistent results in Python but with a slower frame rate (I assume less than 10 FPS). A test that you might try is to increase the voxel size such that the pipeline need to process fewer points. You might obtain less accurate results, but they should be at least consistent over different runs, please let me know how that goes.

Shahrullo, 2023-05-04:
@tizianoGuadagnino Thanks for the response. Indeed, after increasing the voxel size, Python ran faster, but as I increase the voxel size, the results are getting worse. And the results are consistent. 

I have some extra questions if you don't mind: 

1) How to avoid a frame-dropping problem in ROS?
2) Is there other ways to increase FPS (frame rate) except modifying voxel size?
3) In each run, FPS is high at the beginning and dropping significantly as the process goes on. What would be the reasons? 
4) Is that possible to run the kiss-icp on GPU for speed acceleration?

tizianoGuadagnino, 2023-05-05:
In principle, there are two ways to make the pipeline run faster so avoid the frame dropping on the ROS side. You can either increase the _voxel_size_, as already mentioned or decrease the _max_range_ so that the local map in which KISS is searching for correspondences during registration is smaller. This last point also explains the "FPS" dropping you might experience; the local map became too dense over time, and even with voxel hashing, searching for correspondences is too expensive. In general, _max_range_ is the main (in most cases, the only) parameter that with tune when the default configuration does not give us the results that we want.

Regarding the GPU implementation of KISS, at the current state, we do not plan to provide such a backend for computation.

I hope I have replied to all your questions :smile:

Shahrullo, 2023-05-08:
@tizianoGuadagnino Thank you so much for your detailed response. Indeed, decreasing _max_range_ was really helpful and it increased the speed (of course by deteriorating the outcome :laughing:). I will do more tweaking of _voxel_size_ and _max_range_. I appreciate the time and effort you took to provide such a clear explanation.

closed 2023-05-08 by Shahrullo