PRBonn/kiss-icp #155: Is Linear Velocity considered in DeSkewScan method?

BurhanMuhyiddin, 2023-05-12:
Hi all,

First of all, I am very thankful for your beautiful work. I have read your paper which was very clear and informative. Now I am trying to study the source code. However, I stuck in the **_DeSkewScan_** method, which accepts _**skewed points, relative timestamps, start and final poses**_ as inputs and returns **_deskewed pointcloud_**. In the paper, you got relative pose between starting and final poses and calculated translational and rotational velocities. Then, based on the assumption of constant velocity model you have deskewed the pointcloud. Below I attached formulas from paper for being more precise.

[image]
[image]

However, in the source code on method **_DeSkewScan_** only rotational velocity has been taken into account. May I know why? Or am I missing the linear velocity part? Also, is **_mid_pose_timestamp_** corresponds to **_delta_t_**, namely lidar acquisition time?

[image]

Thank you very much for your effort and time.

YuePanEdward, 2023-05-15:
Here the SE3 pose is used, so I think both the translation (linear velocity) and rotation (rotational velocity) are considered.

BurhanMuhyiddin, 2023-05-16:
@YuePanEdward Thank you for your answer.  Indeed I am still surprised that how I didn't see that in the code SE3 pose is used :)

closed 2023-05-16 by BurhanMuhyiddin