PRBonn/kiss-icp #157: Points timestamp reading error

YibinWu, 2023-05-12:
https://github.com/PRBonn/kiss-icp/blob/254264719d34bb5df1ec84a069b117b28dfec812/ros/ros1/Utils.hpp#L86
Hey @nachovizzo , I found a possible issue here.  Should the datatype of the "time" field in the Velodyne point clouds message be float instead of double?  
I tried this [dataset](https://github.com/SJTU-ViSYS/M2DGR) where a Velodyne VLP-32C LiDAR was used. It cannot extract the timestamps correctly unless I change double to float.

nachovizzo, 2023-05-19:
Hi @YibinWu , sorry for the late reply! My current workload is hell!

Thanks for pointing this out. I will try the dataset next week. Although I have the feeling that it might be tricky to solve, I will find a way to also support `float` timestamps.

Question: Have you tried changed that type and was it working?

YibinWu, 2023-05-19:
Hey, no worries!
Yes, I tried. It worked only with **float** on that dataset. It corrupted if I used **double**. I feel it may depend on the model of the LiDAR sensor. For example, for Velodyne LiDAR msg, the timestamps are stored in the "time" field in **float**.
Here is the datatype definition in the Velodyne driver source code. 
https://github.com/ros-drivers/velodyne/blob/4dc95acbdfa31bacc1844be8d6a4f92517706ca9/velodyne_pcl/include/velodyne_pcl/point_types.h#L48
And I also saw other open-source code reading the timestamps of Velodyne LiDAR point cloud with **float**. FYI.
https://github.com/hku-mars/FAST_LIO/blob/0dbfc6e1cb0bd7fc209bee6de83c01b49ac9398a/src/preprocess.h#L37

Lynda549, 2023-06-12:
hello,
For my case,i am using Ouster LiDAR and the timestamps  i got  are bad and not usable. I would like  to have some  information regarding this issue.
Thank you.

nachovizzo, 2023-06-12:
@Lynda549 without acess the data is impossible to know what's going on

Lynda549, 2023-06-12:
hello,
Thanks for your answer,here is the data.
[CrucialX8_poses_tum.txt](https://github.com/PRBonn/kiss-icp/files/11723677/CrucialX8_poses_tum.txt)

nachovizzo, 2023-06-12:
That's not the data, only the poses.

Lynda549, 2023-06-12:
Sorry,
Do you want a pcap?

nachovizzo, 2023-06-12:
Yes

Lynda549, 2023-06-15:
hello,
Sorry for my late answer, you will ind attached my pcap bellow. I would like to know the problem of timestamps and the drift that i have on trajectoy on Z axex.
Thank you so much.
[2023-06-12-09-18-35_OS-1-64-122138000260-1024x10.zip](https://github.com/PRBonn/kiss-icp/files/11755641/2023-06-12-09-18-35_OS-1-64-122138000260-1024x10.zip)

nachovizzo, 2023-06-15:
@Lynda549 Please be more considerate. That file is just a json file. Open a new issue with the entire description of your problem and provide data. Do not comment on this issue. Please do exactly like this, or we will need to ban you from the repository.

Lynda549, 2023-06-23:
hello,
Im so sorry, they were stored in the same file and i choose the wrong one.I have a problem on sending my pcap it has 26 GB this platform doesn't support pcap file and to compress it is too much .Do you have any any suggest for me to send you the data?
Thank you.
Cordially