PRBonn/kiss-icp #161: Support RGB & Intensity fields of a PointCloud

roncapat, 2023-05-18:
I saw, looking at ROS 2 wrapper, that only fields X,Y,Z are copied from PointCloud2 Message to Eigen structure. This of course mean that no other data is used for registration. Is there a rationale for this/are there plans to evolve the package to integrate those info?

nachovizzo, 2023-05-19:
We deliberately ignore all other fields without the points or timestamp information. The reason is that the registration pipeline does not need the other fields at all, which would complicate the implementation for (apparently) no reason.

It's annoying if you want to visualize fancy pointclouds with intensity, etc. But you can create an easy "mapping" node that just subscribes to the cloud topic and creates one. The other trick I do is to visualize the original cloud and increase the "decay time" on RVIZ, this will create a fake, visualization-only, pointcloud map with all the original fields from the msg.

Do you have any other application in mind?

roncapat, 2023-05-19:
Well, my question was related more to the registration itself. I know there are registration strategies that take into account also the color (maybe just the hue value) of each 3d point, and even ones based on intensity values. As an example, https://pointclouds.org/documentation/classpcl_1_1_generalized_iterative_closest_point6_d.html#details but I haven't checked whether the intrinsics of the algorithm proposed there are too different from this ICP implementation.

nachovizzo, 2023-05-19:
I see there is also the ColorizedICP in Open3D.

In our experience, this does not improve much the registration results, and KISS-ICP is all about finding the bare minimum working system. Adding colors to the registration pipeline would propagate complexity form the data-loading scripts, up to the optimization backend we use to align the clouds. So I guess this is more like an application for a specific sensor/dataset.

Thanks for asking anyways :)

closed 2023-05-19 by nachovizzo

roncapat, 2023-05-19:
Thank you for clarifications :)