PRBonn/kiss-icp #162: Make ROS2 OdometryServer composable

roncapat, 2023-05-18:
It shall be easy to transform OdometryServer in a ROS2 component. Is this way, you could provide:

- a standalone node (as it is now)
- a component

... at the same time.

Moreover, the component would benefit of much higher transfer rate of PointClouds messages thanks to intra-process communication.

Would you like a patch to provide this?

nachovizzo, 2023-05-19:
@roncapat Thanks for the suggestion. As I read this, I get to know about components in ROS 2. So, I'm literally blind to this topic. I'm more than willing to help review a merge request about this.

Would you be interested in doing so? If not, could you provide some pointers on where to start and what example to look in order to achieve this?

Thanks!

roncapat, 2023-05-19:
If you'd like to know more about composition, a paper has been released this week :)
https://arxiv.org/abs/2305.09933

nachovizzo, 2023-05-19:
Just saw the post yesterday on LinkedIn. Amazing works from Steve as usual